CTCaer/hekate #883: Emummc Restored partition - existing sd storage issue

dnsBlah, 2023-08-27:
Right so literally everywhere I go and ask, there is no proper answer.
I have problems with the sd storage when migrating emummc partition.
I just make a emummc rawgpp backup from Hekate.
Incl. boot0/1 files (because people tell to do that, works without anyways..)

I prepared the new SD through Hekate (also tried to use tegraexplorer to prepare)
Copied everything over and restore boot0/1 and rawgpp.
Emummc launches fine, don't get me wrong, the restore does work..
All applications that where installed to the internal storage of emummc works fine.
But all applications that where installed to the "sd storage" (folder sd:/emuMMC/RAW1/Nintendo/) are unable to launch..
However the folder size is still the same as on the source SD.

Is this known? And is this solvable or is there a workaround with some other tools?

CTCaer, 2023-08-28:
If you are using a resized emummc, it's normal. Resized emummc wipes USER partition.
If that was not the issue then you didn't copy the paired nintendo folder.
Do a normal emmc backup and also copy properly its nintendo folder.

closed 2023-08-28 by CTCaer

dnsBlah, 2023-09-01:
I'm not sure I understand what you mean by resized emummc, I'm just using the 29/full emummc partition.

- Create emummc
- Install application to "internal storage"
- Install application to "sd storage"
- make emummc backup
- copy all data to PC
- wipe SD
- prepare SD with partition (either hekate, or tegraexplorer)
- copy all data to SD
- restore emummc
* run archbit fix, makes no change (before or after first time launch)
* run fix raw, makes no change (before or after first time launch)
= application on "internal storage" works fine
= application on "sd storage" does not work at all, needs to be reinstalled

CTCaer, 2023-09-04:
That's because you didn't copy the nintendo folder in the appropriate subfolder in emuMMC folder.
As I said before:
>If that was not the issue then you didn't copy the paired nintendo folder.