CTCaer/hekate #885: Hekate freezes on Hekate IPL logo screen at boot - hardware issue but need help

dhattey, 2023-08-28:
I am pretty sure this is a hardware issue but the folks on ReSwitched discord advised i post it here.
I use Hekate a lot - I repair switches and outfit the unpatched ones with CFW - THANKS by the way!
Switch I'm working on now boots HOS fine, docks, reads games, attaches to wireless and Bluetooth, not banned.
Freezes as noted above when trying to boot Hekate using RCM-1.  First change was loaded 6.0.6 onto RCM as payload.bin.
Tried different SD, tried full new setup of SD, tried different SD daughter board - same result.
At suggestion on ReSwitched, tried boot without SD and simple UI comes up fine.  And I plugged the SD in after that and used console info on simple UI to test SD and it passed - showed all the info as expected with a Current Rate of 104Mbps
Tried putting Fusee.bin on the RCM as payload.bin, found I needed to boot that SD into Hekate on a different switch and fix archive bits, but then it booted to Atmosphere fine when I injected Fusee!

So I know this is not a Hekate issue - but what is Hekate doing when it is displaying the Hekate IPL screen initially?  Knowing what could hang it there will give me some idea where to look to find the hardware issue.
THANKS in advance, and thanks for all you do for the community!
Dave

urherenow, 2023-08-28:
Did you search the closed issues here? I forgot specifics, but on certain clones/revisions of dongles, they don't handle hekate well after a specific version. Forgot what version that was, but looking at one of the issues will provide that info.

Anyway, the version on the dongle doesn't matter a lot, because if there is a newer update.bin on the sd, it will load from sd regardless. Hope that helps.

CTCaer, 2023-08-28:
Check https://github.com/CTCaer/hekate/issues/877#issuecomment-1684467585

dhattey, 2023-08-28:
I didn't see anything that stuck out around freezes, had tried deleting nyx.bin but that didn't work.
Thanks for directing me, CTCaer, just opened the .ini and bpmpclock was set to 2, i upped it to 3 but still frozen.  should i try a different value?

dhattey, 2023-08-28:
Just bumped it to 5, still freezes.

dhattey, 2023-08-28:
I just noticed the comment on random premade stuff - i use pure downloads from your repository and the Atmosphere repository.  I have edited the .ini file for Hekate to get the launch parameters and pictures i like but thats it - no EMUMMC or piracy.  But one of my first attempts to fix this one was to take a new SD card, fresh FAT32 format, and loaded pure zip extracts from the two GitHub repositories and it does the same thing.

CTCaer, 2023-08-28:
There's no 5 in that.
If 3 freezes then it's SD card issue. Fake or problematic connector mobo.
If not then ram issue. You probably cracked a solder ball with no or improper preheating.

Send a pm in discord. You can find my tag in retronx one.

CTCaer, 2023-08-29:
Update:
Ram is fine which is good. No reflow of SoC is needed.
It seems to be a sd card issue (probably minerva is read in a corrupted state in Nyx sd init and hangs the whole thing).

Now though, the sd fails to init a 2nd time completely.
On the other hand, that's somewhat consistent behavior with before, since hekate worked fine before but Nyx was hanging (hekate logo screen is Nyx init).
Since it's a hw issue, there won't be any further update on the situation here.