CTCaer/hekate #888: [Enhancement] - Battery cycles

Issue-maker-9000, 2023-09-11:
For all the useful battery info it gives, it doesn't display the battery cycles which I think can be quite useful.

masagrator, 2023-09-12:
I am quite curious because when last time I have tested this readings seemed unreliable.

Like first time when I was reading it, that was 250, after few hours of charging I got 251 and an hour later all of a sudden it's 257.

CTCaer, 2023-09-12:
@Issue-maker-9000 
That won't be implemented since HOS clears it when it exceeds more than 2 charge cycles and adds it to its own cycle counter.
So there's nothing to show if it will always be less than 3 cycles (300%).

@masagrator 
Did you read the description on how it works? It is the cycle register.
It saves cycle percentage not a pure number.
65535% means 655.35 cycles (251%: 2.51, 257%: 2.57).
But anyway, it's useless since hos clears it.

closed 2023-09-12 by CTCaer

masagrator, 2023-09-12:
> Did you read the description on how it works? It is the cycle register.

Yes, but since it max limitation is 655 cycles and it resets, it is not most people are searching for.

CTCaer, 2023-09-12:
Again, the issue is not the reg size limitation and it doesn't reset on its own.
I already said why it's useless and not what people are searching for.
HOS sets it to 0.96 when it exceeds 2.96 charge cycles.

Issue-maker-9000, 2023-09-12:
> @Issue-maker-9000 That won't be implemented since HOS clears it when it exceeds more than 2 charge cycles and adds it to its own cycle counter. So there's nothing to show if it will always be less than 3 cycles (300%).

Ah that really sucks. makes sense now why no homebrew shows cyclecount. Thanks for the answer and hekate ofcourse!

CTCaer, 2023-09-12:
There's a homebrew for that since 3-4 years ago ([BatteryInfo.nro](https://cdn.discordapp.com/attachments/458344156184707077/688849543097483414/BatteryInfo.nro)). And any homebrew can show it.
HOS saves it first before clearing the actual hw and that data is exposed.

Issue-maker-9000, 2023-09-12:
> There's a homebrew for that since 3-4 years ago ([BatteryInfo.nro](https://cdn.discordapp.com/attachments/458344156184707077/688849543097483414/BatteryInfo.nro)). And any homebrew can show it. HOS **saves** it first before clearing the actual hw and that data is exposed.

Ohh I misunderstood, makes sense now so hekate reads it out directly from the battery management chip or whatever it uses. While that hb application gets it from os itself. Correct?

I honestly missed that hb app, thanks for the link! Seem to have quite a bit of wear for the cycles it has gone through, 93% health on 235 cycles while ninty rates it for 80% health on 800 cycles. Assuming it is accurate of course

masagrator, 2023-09-12:
> Assuming it is accurate of course

This is saved to nand, so every backup restoration or switching between emummc and sysmmc regularly will cause this reading to being unreliable.

CTCaer, 2023-09-13:
Yeah the only 100% reliable way is to only use one mode and never restore.
On the other hand it can be manually increased/fixed, assuming someone notes them down.
Best way is to note down the counter when a copy is done. Each increment on both sys and emu counts.