CTCaer/hekate #889: [Enhancements] Sleep Mode in Hekate

mickael28, 2023-09-13:
Hi guys, I read some mentions in the project and on different forums but after a few hours researching and trying things I didn't manage to find an answer or clarify the progress of this.

I wanted to have a dedicated SD card for loading Android on my Switch, as such it'll mainly have a large Android partition, and then I'll have a different card with the CFW-emuMMC partition.

I had the idea to reboot into Hekate when swapping cards, leaving the console in Sleep Mode inside Hekate (so that it'll last a pretty long time without having to bring all the cables, RCM loader, etc) and then be able to easily put whatever card I wanted to use next time.

The issue I've been having is that on the SD card with Android only, I cannot see a way to put the console into Sleep Mode.

Is this feature planned to be included in any of the upcoming Hekate releases?

Cheers!

CTCaer, 2023-09-13:
I wont add that for the whole reason that it cant be done like that in mariko (many important registers get completely locked).
You can use l4t linux and deep sleep. That will give you the highest sleep time. And then reboot and change card in hekate.

closed 2023-09-13 by CTCaer

mickael28, 2023-09-14:
Thanks for the quick comment, at least it's clear now, and for the tool.

Mainly for future reference and to think about the possibilities, would you know how these sleep times might compare with each other?
* CFW-emuMMC => and click `sleep`
* Android 11 => and just tapping the power button? (thinking this will put the console to `sleep` as well?)
*  l4t linux => the `deep sleep` you mentioned (I don't know much about that setup or deep sleep atm)

CTCaer, 2023-09-14:
L4T Linux or Android (assuming android fixed the random hangs on wake up or whatever).
Both enter sleep via power button. Linux one is forced, nothing can stop it and be in a awake state with screen off.
On linux if screen is off then it's in deep sleep.

HOS or HOS with CFW is the worst even with airplane mode or custom sync settings.
And is even worse on mariko thanks to a bug on the pcie drivers for t210b01 that HOS has (I fixed that on L4T).

mickael28, 2023-09-14:
I'll try from Android for now then until I have a look at that L4T Linux you mentioned (maybe on a different SD card to see how that works)

Thanks so much, really helpful!