CTCaer/hekate #890: Hekate detects Stick movement on Switch lite with hall effect sticks

Clydefrosch22, 2023-10-02:
On a new switch lite with picofly chip and new gulikit sticks, Hekate seems to detects movement making the cursor appear and wander typically to the bottom right of the screen.
It's the same issue described here https://github.com/CTCaer/hekate/issues/847

The sticks are calibrated and show no sign of drift or ghost inputs whatsoever during normal switch play or in the calibration menu, they are not being touched during startup to either.

Anytime i reload hekate with the reload button, the cursor will move differently. Its either straight right, hard down and right or slow down and slow right.

Is it possible the calibration hekate performs at startup doesn't play well with the hall sensor sticks? 

It is not at all a big deal and if there is no solution or no incentive to look into it, that's fine too.

CTCaer, 2023-10-11:
>Is it possible the calibration hekate performs at startup doesn't play well with the hall sensor sticks?

As you guessed hekate uses its own dynamic calibration and is done when they are detected and joycon init finishes, so it's fully controlled.

So it could be. If you want to test a mitigation for that, find my discord tag in retronx and pm me.

redbeard1083, 2023-10-12:
For whatever it is worth I have the exact same setup and see the same behavior.

ermonas, 2023-10-13:
Same setup, same issue. I’m willing to help debug a possible solution, if you need any help. Thanks!

CTCaer, 2023-10-15:
Test:
[nyx.zip](https://github.com/CTCaer/hekate/files/12909212/nyx.zip)

redbeard1083, 2023-10-15:
Issue resolved for me.  Thank you!

DiamondYuan, 2023-10-15:
Issue resolved for me too.

ermonas, 2023-10-16:
I can confirm the fix works too! Thanks!

HexNoob2, 2023-12-02:
I tried this zip and it also fixed my issue:  Left stick always going down at about 50% in hekate.  Thank you.

harvestry-of-ghosts, 2023-12-15:
> Test: [nyx.zip](https://github.com/CTCaer/hekate/files/12909212/nyx.zip)

I do not have Hali sticks and this issue was happening to me with the latest nyx. Tried this and it immediately went away. Can you please explain what changes were made to fix it? Some people in #847 said it is 100% related to stick drift. Is that accurate? As I experience zero drifting during game play and zero drifing during the iitial loading of Hekate. It ONLY happens when reloading it. Is the fix going to be in 1.5.7?

CTCaer, 2024-02-22:
Fixed in Nyx 1.6.0

closed 2024-02-22 by CTCaer

mbuna1992, 2025-02-17:
Hi sorry have this issue, where do i put the nyx file?

Leerz, 2025-02-18:
> Hi sorry have this issue, where do i put the nyx file?

roo:\bootloader\sys\nyx.bin