CTCaer/hekate #894: Optimized Firmware 17.0.0 + Atmls 1.6.0 support

Arch9SK7, 2023-10-12:
I'm pretty certain you are already working on it. But I thought I would open this just in case.

RyabDev, 2023-10-12:
Yeah it seems at the moment it is completely non-functional. Loads to a black screen or atmosphere goes to kernel panic.

Visualwit, 2023-10-12:
What is taking so long?

RyabDev, 2023-10-12:
> What is taking so long?

This stuff is done in their free time. Give them time.

Kinokoda, 2023-10-12:
> What is taking so long?

patience is a wonderful thing, please obtain some. :)

impeeza, 2023-10-12:
> What is taking so long?

Aleays you can ipen a development IDE and start to code by yourself.
The cide is open, and the developers had invested lots and lost of their free time and knowledge on the this, then give to all us for free.

So please, give some credit and if you like to get a very quick development you can pay to somebody a lot of money, code all the needed changes by yourself. or grateful wait.

Please by respectful

Kinokoda, 2023-10-12:
> Aleays you can ipen a development IDE and start to code by yourself. The cide is open, and the developers had invested lots and lost of their free time and knowledge on the this, then give to all us [...]

oh huh its impeeza. heyo, thanks for helping peeps out over on gbatemp, you deserve the pog award.

SciresM, 2023-10-12:
let's not shit up CTCaer's issues, guys. He works really hard, and I know for a fact that he's already put a bunch of work into 17.0.0 stuff.

ThatOneOcto, 2023-10-13:
> What is taking so long?

Is this real?

GelidaAvis, 2023-10-13:
> It's insane looking at the previous releases just how fast these devs are at updating the projects after a new switch firmware release, just over 24hours and AMS has a pre-release that works.
> 
> 
> 
> also, hekate worked fine for me on a fresh unit. I did get a kernal panic once, but after running again worked fine. I'm going to hold off on booting into AMS fully though just in case.

how were you able to boot? once i installed ams 1.6.0 it won't boot, just gives me a panic.

TwistedTempest, 2023-10-13:
> how were you able to boot? once i installed ams 1.6.0 it won't boot, just gives me a panic.

I had the same issue, delete mission control or whatever folder under atmosphere/contents is failing. According to SciresM there is a possibility for soft bricks, but it could just be something not up to date

Vinfall, 2023-10-13:
Read 

> how were you able to boot? once i installed ams 1.6.0 it won't boot, just gives me a panic.

Read carefully, you get a panic on first boot. But later on it should work fine. If not, try load payloads/fuse.bin instead of selecting nyx launch option. Note this would launch Atmosphere directly and thus ignores launch configs set in `hekate_ipl.ini`.

CTCaer, 2023-10-13:
https://github.com/CTCaer/hekate/releases/latest

closed 2023-10-13 by CTCaer