CTCaer/hekate #897: [Enhancement] Allowing Keyboard Input In Recover Mode

Jazztache, 2023-10-26:
Hello Hekate Team!

I am having an issue where, due to my volume buttons being broken, I cannot navigate the menus correctly nor proceed with installing custom firmware.
If recovery mode supports keyboard input, would it be out of the question to allow USB C keyboard input? This functionality would use the arrow keys, with Enter to confirm.
Thankyou!

urherenow, 2023-10-26:
Which part of hekate uses volume buttons, exactly? nyx fully utilizes the touch screen, so your question makes little sense. If you're talking about using something like lockpick_rcm, then hekate has nothing to do with whether or not THAT software can use something other than volume buttons. It only launches the payload, then everything is that payload's job.

masagrator, 2023-10-26:
Sounds like you didn't install nyx. Install nyx that is included with Hekate already for a long time, and you will be able to use touch screen.

closed 2023-10-29 by CTCaer

Jazztache, 2023-10-29:
Ah apologies, seems I mistook Hekate for something else. I was referring to the TUI payload app. 
I might revisit this if I find any issues with Docked mode in Nyx. I think this may be an issue with fusee_gelee, so I'll send this their way instead. 
Thanks for clearing that up.

Jazztache, 2023-10-29:
Ok, I will check my Nyx installation. I'll have to attempt a re-install, since Hekate isn't detecting Nyx. Didn't realise Nyx was the frontend and should load immediately.