CTCaer/hekate #899: Samsung EVO Plus MicroSD cards not working

Spoonzz6669, 2023-11-01:
Just here out of curiosity. I have a switch V2 with a picofly. 3 of my 512gb Samsung Evo plus microSD cards 9/10 times will display failed to init SD card. Tried my friend's, my old one and bought a new one 2 days ago.
 
Sometimes it will show on the picofly logo screen or it actually loads the Hekate logo and displays the text on the top of the screen (vertical orientation).

These do work in OFW no problem and always shows up in data management and I've tried installing and playing games via eshop with no issues.

What works (flawlessly):
32gb verbatim
128gb SanDisk extreme
512gb SanDisk extreme 

I thought it was the connector until I replaced the board where the same problem existed so I decided to buy a 512gb SanDisk extreme today.

All cards are genuine and purchased as legit entertainment goods store here in my country. They are also formatted the same - FAT32, 32K clusters

What do you think can be the cause? I'm going to assume it's picofly itself but in case it's not, I thought I'd ask here hoping to be able to use it one day.

damex, 2023-11-06:
are you sure they're genuine Samsung cards? where did you get them? there is lots of counterfeit products on a market. 

are they of a new kind with blue letters and white background? or it is the old one with orange and white background?

you can try cleaning contacts with qtip and 90% alcohol

i am am personally using older orange+white samsung evo plus 512gb sd card and it works flawlessly on my 'patched' v1 switch (would that make it v1.5?).

Spoonzz6669, 2023-11-06:
> are you sure they're genuine Samsung cards? where did you get them? there is lots of counterfeit products on a market.
> 
> are they of a new kind with blue letters and white background? or it is the old one with orange and white background?
> 

They were purchased from JB Hi-Fi here in Australia. Massive nationwide entertainment goods store over here. They work on my OLED + Hwfly fine. They are definitely genuine.

Yeah they are the blue letters on a white background. Just found it a bit strange as to why the picofly can't read it.

BeckysFootSlave, 2023-11-06:
Do you use the latest firmware 2.7.3 on it?
https://github.com/Ansem-SoD/Picofly

Spoonzz6669, 2023-11-06:
> Do you use the latest firmware 2.7.3 on it? https://github.com/Ansem-SoD/Picofly

Yep I am but I believe I got the firmware from another GitHub link and not the one in your comment. Also to add, the Samsung's work on everything else no issues but on my V2 switch with picofly.

dkorunic, 2023-12-06:
Not sure if this helps, but on my Mariko V2 unit with PicoFly with 2.75 firmware Samsung EVO Select 512GB SD card (from Amazon) works without any issues.

CTCaer, 2023-12-30:
Screenshot from console info -> microsd should always be provided in such cases when you can enter Nyx.

closed 2024-02-09 by CTCaer

mxarment, 2024-03-27:
Having the same problem on a friends switch OLED with the picofly,

Ive tried installing the HATS pack in the card in case some corrupted file but it always show the error "_failed to init sd card_"
Also, trying that same pack on another SD card (a sandisk one) works perfectly.

I dont know what could it be but it happened from one day to another without an apparent reason, and what is more curious, I personally have a switch lite with the same chip and the exact same card bought on the same place, and it still works for me.

I dont know if it could be a manufacturing error but makes no sense because the PC can read the card and its files perfectly.

I would appreciate any ideas in case someone has some clue about what can be happening

(Attached photos, one from my lite with the card info and the other from the oled displaying the error)

[image]
[image]

AnalogMan151, 2024-03-28:
I'm also having this issue. 512GB Samsung EVO failing with both Picofly and Hekate. If I use another SD card to boot Hekate and then insert the Samsung card, it gives a failed to init message after reloading. Works perfectly fine in HOS.

Leerz, 2025-04-21:
not exactly a necro but just for a thought, it's an sdloader issue 

if you change sdcard brands it doesn't have this issue, it seems picofly firmware likes sandisk alot. even if HOS supports samsung.
we know this isn't a modchip support. so i think it shouldn't be raised here.