CTCaer/hekate #901: [Feature request] autoboot on dismount

zemixe, 2023-11-06:
The suggestion is as follows:

Example key:
`autoboot_on_dismount=1`

If possible not used when accessing UMS the traditional way in Nyx but used in the case where the user rebooted directly to UMS, as soon as the volume is dismounted hekate immediately boots the user's autoboot selection provided they have one.

The purpose is for rapid reboots if doing testing that involves constant rebooting to UMS to add/remove/replace files, or if the user 9 times out of 10 is rebooting to hekate UMS mode directly and wants to speed up the experience of returning to their OS.

CTCaer, 2023-12-30:
I've done that procedure tens of thousands of times and I don't think it guarantees it as a feature, just for saving 2-3s.
(Booting to a boot entry or rebooting for autoboot to take is 3 taps away)

closed 2023-12-30 by CTCaer

ranieryfialho, 2023-12-30:
***@***.***#W% as 2 We Z Aww~wAA4

zemixe, 2023-12-30:
> I've done that procedure tens of thousands of times and I don't think it guarantees it as a feature, just for saving 2-3s. (Booting to a boot entry or rebooting for autoboot to take is 3 taps away)

It's ok I'll just wait a year or so for you to add when you realize it's a good idea, like with that issue: https://github.com/CTCaer/hekate/issues/654

(this is a just a little joke before anyone else comments!)

CTCaer, 2023-12-30:
Well that one happened mostly because at one point I refactored the whole boot wait with the addition of the ticker which made it super simple to implement. And it got implemented eventually because it was asked many times.

I still find it useless. Features that are gonna be used by 0.0001% of the userbase (and I won't be using them) are "features" for the shake of being features. Something like a 90s/00s windows app.

You never know though, maybe it will be added. Boredom sometimes is a sucker, especially if feature popularity is still not there.

zemixe, 2023-12-30:
No problem CTCaer. I personally like features that speed up actions that are performed repetitively. I appreciate that it's niche, if even you wouldn't find it useful then it must be really niche!