CTCaer/hekate #905: emuMMC RAW GPP & BOOT0/1 failed to backup

observeroftime01, 2023-11-25:
Using Hekate 6.0.7.

I recently migrated SD cards following this guide [here](https://nh-server.github.io/switch-guide/extras/transfer_sd/) (the partition based emuNAND steps, because that's what I am using). 

Now when I attempt to make a backup of my emuNAND, I am get the following errors:

[image]

[image]

It fails after sitting on "checking for free space" for about a minute. This is not normal behaviour, because I used the exact same steps in the guide to make the previous backup which I then (successfully) restored to the new SD card. The SD card in question: 

[image]

It is a legitimate card I tested with hw2test. I should note that it takes a good 30 seconds to display the "Found FAT volume" information panel on the bottom to show up there. My switch works perfectly fine, the partition-based emuNAND is good and working. The "emuMMC Manage" page tells me that it is enabled, starting sector 0x37F26000. I'm not sure where else to check the partition info on the emuNAND, but here's the following information:

[image]

What steps do I have to take to back up my partition-based emuMMC (for future migration to a new card, or just for general backup purposes)? Is there something I am doing wrong? Virtually the same steps have worked for me before, the only thing that's different is that I am using a 512GB card I formatted with TegraExplorer prior to restoring the backup (and I used hekate 6.0.5 to create the initial backup, but that's likely unrelated).

urherenow, 2023-11-25:
To manually backup a partition based, you can just copy the partition with a tool such as win32diskimager (you'll need to go into disk manager in Windows to assign a drive letter to that partition, but do NOT format or let Windows "fix" the drive). For help manually restoring it and making it work on the new card, please search or just start a new thread on gbatemp or discord. But really, after getting your backup, I think you can easily search for tutorials on converting from partition to file based. Backup is ALWAYS the first step though.

observeroftime01, 2023-11-25:
Thanks for the quick reply. The issue is that I am unable to make a backup of the emuNAND image using hekate in the first place (see the errors above). When trying to back up that partition-based emuNAND, I get the errors that no partitions are found (I know they are there and can see them using gparted / diskimager). I have already copied everything off the card using the UMS function, I just need the emuNAND backup for its contents. The very same steps have worked for me before, and are now throwing the above exceptions. It is worth noting that I have done these steps, with success, many times in the past  and have never seen this error before. Am I not supposed to use the backup function within hekate for backing up the full emuNAND? I know that I can use other tools, but shouldn't hekate work, since the functionality is there and all? I am not trying to restore anything, I'm talking about the creation process of the rawnand.xx & BOOT0/1 files that fails for me (what I call the backup process). Maybe I didn't explain my issue clearly enough. I can manage the restore process fine, but thank you for the suggestion. Since I am running into errors creating the BACKUP using hekate, I thought I'd report this unusual behavior here, sorry if this is the wrong place. Anyhow, if you feel this error with the backup process is not a hekate issue and is instead a case of user error, feel free to close the issue and I will see myself out (though I am going to be still left wondering what to do about the issue with hekate).Thank you for your time and effort 👍.

CTCaer, 2023-12-12:
Just create a new emummc and then restore.
I probably check for existing emummc in order to allow restoring.
I might change that in next version but for now just create a new one and restore.

CTCaer, 2023-12-30:
Reopening, since as I stated, I might add support for restoring on an empty partition.

CTCaer, 2023-12-30:
Ok by reading more carefully the issue is not restore based issue at all.
Also I only validate one thing on restoring so empty partition restoring is fine.

The checks for backup are simple. 0xE0 for partition type and a GPT at a specific offset (0xC001).
Generally, don't ask support if not preparing the sd card via hekate for such things.
I can't validate or have time to police what others do, especially if they don't tell me.

The specifications of all that were set in stone back in mid-2019. Anything else that does not follow them is out of spec.

Lastly, for anyone else, since I doubt op will answer, always take a screenshot of MBR pop up when entering sd partition before posting in issues, so I can know the actual layout of the card.

closed 2023-12-30 by CTCaer