CTCaer/hekate #906: Is there a way to make it recognize 8GB of RAM?

bulzipke, 2023-11-26:
Hi, I saw the issue [this](https://github.com/CTCaer/hekate/issues/878).
I hope this is not a question that is bothering you.

I changed the RAM of the hardware to K4UBE3D4AA-MGCL x 2 = 8GB.
And if I boot from hekate v6.0.7, it will boot normally like the screen below.
[image]
It booted successfully, but it's still showing as 4GB.

https://github.com/CTCaer/hekate/blob/e84367e30218e22123cc1612938fd8876f64ee8c/bdk/soc/fuse.c#L89C1-L89C34
I saw that my dramid used by hekate is 28, and if I just do a return 28 here, it is recognized as normal like below.
[image]

Do I need to hardcode this and use a private build every time in the future update?
Or, if not, I wonder if there's a way to make it recognize dramid separately in the config file.

Thanks as always for your hard work.

CTCaer, 2023-12-12:
Yeah you need to hardcode that and compile your personal build everytime.
A file based configuration can't be used because at that point ram is initialized and utilized. (And you can't just change the ram config.)
The good thing is that hekate is not updated for 0 reason just to get updated. I only release when it's needed.
So that would not be often enough to get annoying.

closed 2023-12-12 by CTCaer

fesily, 2024-07-29:
Sorry to revert to this issue, but do you have a tool for how to permanently blow a fuse?

impeeza, 2024-07-29:
> Sorry to revert to this issue, but do you have a tool for how to permanently blow a fuse?

What?  Why?  This is not a support site, please go to Discord or GBATemp, or read a guide about basic Ninty Switch concepts.

fesily, 2024-07-30:
I can find the programme_fuse_otm4_t210_t210b0.bin, but following this video on YouTube [Switch OLED Pro - 8GB RAM / 512GB NAND Upgrade - Full Guide](https://www.youtube.com/watch?v=J4oXmQGZgzU)
This video reminds
[image]
My machine is dramid:27
So I don't know if there is a way to directly modify fuse

urherenow, 2024-07-30:
This is NOT a support forum. If you update your system, and allow it to fully reboot without cfw, with a game cart in the game slot, then every fuse that is supposed to be burned, will be burned. End of discussion.

fesily, 2024-07-30:
> This is NOT a support forum. If you update your system, and allow it to fully reboot without cfw, with a game cart in the game slot, then every fuse that is supposed to be burned, will be burned. End of discussion.

Are we talking about the same thing? I just want to know where on youtube this code can be seen, youtuber said that CTCaer provides this fuse programming. But I can't find the code snippet anywhere on the net.

urherenow, 2024-07-30:
github has nothing to do with Youtube, whatsoever. This space is to report bugs or for collaberation. Ask your questions in a support forum such as discord or gbatemp.