CTCaer/hekate #909: More than an issue a question about partition layout.

impeeza, 2023-12-09:
Hello there, I never had look to close to the partition schema of my cards, recently I migrated to a new SD card, so I did remove all partitions on the recently acquired card, then put that card on the Switch and using NYX's Tools, SD Partition & USB, Partition SD Card I did repartition the Card using The default distribution of 0 emuMMC, 0 Linux and 0 Android (user) partitions leaving HOS (FAT32) at all the space of the card:
[image]

[image]

After all the new partition set are created and the card is ready to be used a 16 MB Un-allocated space was leave at the start of the card:
[image]

Looking trough the issues and documentation of Hekate only find vague mentions to 16 MB something about alignments, and on the Net I found something about «MSR partition» (https://en.wikipedia.org/wiki/Microsoft_Reserved_Partition) but that is referenced to GPT disks.

Just about curiosity, what is the use for that 16 MB allocated space at start of the card?

Thanks a lot for your great work.

urherenow, 2023-12-09:
If you remove it, you can actually brick your card. Leave it alone. From SanDisk:

"“You may leave the 16MB unallocated space as it is. A portion of the total capacity is used to store certain functions including optimizations of the memory that support performance and endurance and therefore is not available for user storage.”

impeeza, 2023-12-09:
Thank you so much for the quick answer, OK then MS called it MSR and is not only for GPT data.

Cool, never think that was needed, more over when the SD Card comes from factory had a full volume (ExFAT) from 0 to full disk. The analogy of the PBC you give I used to think that data was on the SD Card Controller in conjunction with the most used cells and the refresh data.

Luckily I did used Hekate to "Standarize" the card.

closed 2023-12-09 by impeeza

Masamune3210, 2023-12-10:
The MSR and this are not the same thing. The MSR is just a partition Microsoft came up with for their own uses in Windows. It's used as a storage area for anything that would normally be hidden from the system on an MBR-partitioned disk since GPT doesn't allow hidden data on the disk.