CTCaer/hekate #910: weird yellow screen after i update

Danny52407, 2023-12-09:
hello i am very VERY new to modding my switch and i tried updating my hekate after i accidentally updated my console and haven't been able to mod it again afterwards and now my switch is just this bright yellow screen and i am very confused how to fix it please respond as soon as possible to anyone who knows how to fix this issue thank you

Sarutopi21, 2023-12-09:
i guess you have to wait for the new update for hekate since is not ready for the new switch update yet

impeeza, 2023-12-09:
You don't need to wait for anything, the latest version of Firmware today is 17.0.1 and Hekate hekate v6.0.7 full support it. @Danny52407 DOWNLOAD THE LATEST HEKATE FROM RELEASES PAGE (https://github.com/CTCaer/hekate/releases) and extract the files to the root of your SD Card.

If you still have trouble, you can go to https://gbatemp.net/threads/switch-noob-paradise-ask-questions-here.488277/unread or to Discord and ask for help, this space is for bug report not to support.

druka-grey, 2023-12-09:
Hi! Trying to boot HOS 17.0.1 with Hekate 6.0.7 with the updated files on my SD card gives me a yellow screen too. Probably have to wait for an update.

BeckysFootSlave, 2023-12-09:
Don't know what you're both doing?!
Hekate works perfectly!

urherenow, 2023-12-09:
There isn't going to be an update because it works perfectly. Sounds like your hekate_ipl.ini is messed up. Sounds really strange since the last time I saw this problem was when fw 13.0.0 came out and Atmosphere started usin fusee.bin instead of fusee-primary.

impeeza, 2023-12-09:
What happen if you keep pressed the VOL- button while booting?

druka-grey, 2023-12-09:
Whoops my bad, I messed up and used a previous  version of Atmosphere.

@Danny52407 You need to:

1. Download and unzip [the latest Atmosphere](https://github.com/Atmosphere-NX/Atmosphere/releases/download/1.6.2/atmosphere-1.6.2-master-f7bf379cf+hbl-2.4.4+hbmenu-3.6.0.zip) to the root of your SD card,
2. Download and put the latest [fusee.bin](https://github.com/Atmosphere-NX/Atmosphere/releases/download/1.6.2/fusee.bin) to your SD card in `/bootloader/payloads/`

This solved the issue for me.

Danny52407, 2023-12-11:
@druka-grey that actually worked thank you SO much im so sorry i didint respond earlier ive been busy this weekend but thank you so much for your help its greatly apricated i hope you all have a great day thank you all for the help!

closed 2023-12-11 by Danny52407