CTCaer/hekate #912: Issue properly creating emunand on 1.5 TB SD Cards

s3anami, 2023-12-29:
Created a fresh new SD card, brand new emunand each time starting emunand I receive the" Unknown  pkg1 ('')  version hos version not supported" error.
I have taken an existing SD card that works, cloned that to my 1.5 TB SD Card, and manually expanded the partition and this works successfully to run emunand as a work around.

CTCaer, 2023-12-30:
You need to provide proper info about all that.
For starters screenshots from:
- console info -> microsd
And after partitioning via hekate and creating an emummc:
- tools -> partition sd -> the pop up that shows up
- emummc.ini contents

urherenow, 2023-12-30:
Use the latest hekate/atmosphere files. Inject the latest hekate bin if using RCM, and ensure the latest hekate is renamed to payload.bin on the root of your SD. 

Could you seriously not just search for "Unknown pkg1", either here or on gbatemp, instead of creating a new issue?

s3anami, 2023-12-30:
> Use the latest hekate/atmosphere files. Inject the latest hekate bin if using RCM, and ensure the latest hekate is renamed to payload.bin on the root of your SD if using a mod chip.
> 
> Could you seriously not just search for "Unknown pkg1", either here or on gbatemp, instead of creating a new issue?

I am already doing this, so gfy for being an ass.  I already created an emunand  in hekate, so half your steps are like that of someone that didn't even read. The cards are set up identically except being different capacities.

urherenow, 2023-12-30:
Github isn't a support forum, so take it to gbatemp. This are is for reporting real bugs, not for dealing with dumbshits.

CTCaer, 2023-12-30:
@urherenow Did I not warned you before for posting useless comments without even thinking and doing micro management here?
@s3anami Also did you really disregard what I asked in order to reply on that? Why did you even create an issue here then?

s3anami, 2024-01-11:
[image]
[image]
[image]

[emummc]
enabled=1
sector=0xb14a0000
path=emuMMC/RAW1
id=0x31574152
nintendo_path=emuMMC/RAW1/Nintendo

CTCaer, 2024-01-11:
Well now I have the most important things. Which show that the issue is not in creating.

[image]
[image]

I recently fixed an issue in parsing numbers in inis. Send a pm to test that. You can find my tag via retronx discord.

s3anami, 2024-01-12:
i pmed on discord

CTCaer, 2024-01-12:
So booting does not work.
And the thing I fixed (integer overflow during parsing), fixed that also and now boots.

BeckysFootSlave, 2024-01-13:
We also have someone on our Group that has the same Problem!

He set up his 1.5TB Card with all the lastest versions (ams, hekate)
If he boot into CFW (no matter if emuMMC or sysMMC) he gets the same error and nothing boots up!

If he set the command in hekate_ipl.ini from fss0 to payload he can boot into CFW!

@CTCaer what Info do you need from him or what can I tell him?

CTCaer, 2024-01-15:
@BeckysFootSlave 
[hekate_fix.zip](https://github.com/CTCaer/hekate/files/13933661/hekate_fix.zip)
use as update.bin and as payload.bin or injected payload.

BeckysFootSlave, 2024-01-18:
> @BeckysFootSlave [hekate_fix.zip](https://github.com/CTCaer/hekate/files/13933661/hekate_fix.zip) use as update.bin and as payload.bin or injected payload.

May I ask you which files changed in source for that fix?
Would like to compile the german translated hekate fork I have!

THANK you so much!

MartinHHW, 2024-02-15:
> @BeckysFootSlave [hekate_fix.zip](https://github.com/CTCaer/hekate/files/13933661/hekate_fix.zip) use as update.bin and as payload.bin or injected payload.

Hi CTCaer,

I got the same problem on a new 2TB SD Card. Tried your solution with the hekate_fix.zip, but didn't work for me.
Can you do a fix including 2TB cards, too. As I read, it will the max which switch can support. So should be the last time... ;-)
Thank you...

CTCaer, 2024-02-16:
the fix is for 2tb also.
assuming you really used that hekate and you want to qualify as the same issue you need to provide the following screenshots:
- console info -> microsd -> the whole screen
- emuMMC -> the whole screen
- tools -> partition sd -> the pop up that shows up
- sd:/emuMMC/emummc.ini contents

MartinHHW, 2024-02-19:
Hi,

Sorry for my late response, I wasn't at home.
Anyway, I already ordered a 1.5TB card as testing the other one brougt write error in 1 sector.
I'll send it back and be on the safe side then, as 1.5TB seems to work fine...
Hope so at least.

Thanks for your answer and great work!