CTCaer/hekate #913: Hekate exception occurred (LR EB00003E) RESET

zitlalyr2, 2024-01-02:
[image]

Every time I boot up through payload this happens.
I separate issue but when i try to boot through ofw to get past hekate, it’s just a black screen.
Maybe it’s internal damage? I tried replacing the sd reader just in case, but that isn’t the problem.

zitlalyr2, 2024-01-02:
When switching the readers, I also get this screen as well. No other information.
[image]
[image]
Would I just have to update?

impeeza, 2024-01-02:
Could be an hardware damage.

try another SD Card with only Hekate on it  and if the same error, could be a reader or connector damage.

CTCaer, 2024-01-03:
One of your ram chips or the ram solder balls on SoC are broken.
Easily observed by the background artefacts.
[image]
Reflow of the soc or/and ram chips might fix that.

closed 2024-01-03 by CTCaer