CTCaer/hekate #914: Unknown EmcDvfsTimingTableIndex: -1

Calabor-Hoc, 2024-01-03:
Hello folks!
I recently purchased a Nintendo Switch brand new which is a Mariko, I already installed a modchip (the picofly from aliexpress which is ready to install with flex cables). I can enter to Hekate, But when I launch CFW on emuNAND. I get the fatal error: Error: Unknown EmcDvfsTimingTableIndex: -1. Also I notice something weird, I opened console info in hekate and the row that shows the memory says "contact me!" and this tell me maybe this memory is not supported yet? who should I contact? 
[image]

[image]

impeeza, 2024-01-03:
Sadly is more possible you messed up with your NAND, seek for help on discord of forums of installation, like GBATemp.

masagrator, 2024-01-03:
> ###
> **[impeeza](/impeeza) ** commented [Jan 3, 2024](#issuecomment-1874914506)

Nope, recently Nintendo pushed new RAM chips and new Micron ones were expected to arrive at any moment.
Go to RetroNX discord, ping CTCaer in `#switch` channel and provide info about it. To make this faster, if possible take pictures of RAM chips.

closed 2024-01-03 by Calabor-Hoc

Calabor-Hoc, 2024-01-03:
@masagrator You are a savior.

Would Hekate's fss0 method be able to make my console into a dual system(the original unpolluted OFW in SysNand and a CFW on SD card)? A guide link about  Hekate's fss0 would be really helpful.

CTCaer, 2024-01-03:
@Calabor-Hoc Join retronx discord and ping me.

CTCaer, 2024-01-05:
@Calabor-Hoc Hmm did you keep photos of the ram chips during installation?

Calabor-Hoc, 2024-01-06:
@CTCaer  No. Not that I wouldn't bother to open the case and take the photos for you. The chip implant is done by a mechanic in his workshop. I myself know nothing about switch repairmen, and am afraid to screw something If I did it myself.

CTCaer, 2024-01-06:
> @CTCaer No. Not that I wouldn't bother to open the case and take the photos for you. The chip implant is done by a mechanic in his workshop. I myself know nothing about switch repairmen, and am afraid to screw something If I did it myself.

Well checking up on the ram chips requires opening it and then removing the heatsink and the emi shield, which might not be easy if you never done that.
So, can you ask the mechanic if he took video or photos at least?

Calabor-Hoc, 2024-01-06:
I didn't ask the mechanic but I'm pretty sure he doesn't take any photos. He just did this as a part time job(Charge some fee) and knows nothing about hacking switch more than following those online tutorial videos. He never would thought the ram chip would be different for different switch console.....

Anyway, I would message him about the photos and reply here. Your method of "fss0" worked for me perfect at least until now, so I never return to him for any "after-sales service".