CTCaer/hekate #916: missing lp0 sleep lib

superxoi, 2024-01-13:
i bought a BSoD switch lite, changed the emmc to rebuild the nand but the console keep showing error code i could not enter hekate. i have used 2 micro sd cards. the first one was sandisk the console boot and show missing lp0 sleep lib error immediately.  the second was samsung which delay for 3 seconds before showed the error " hang detected in lp0/minerva" . 
please help me. thank in advance.

CTCaer, 2024-01-13:
Show me a photo of the screen with these error messages. Should be as clear as possible (not blurry)

superxoi, 2024-01-13:
[image]
Sr but my phone camera is bad. Bear with me. This one is from my sandisk one. 
[image]

superxoi, 2024-01-13:
When i press a button it shoew this one
[image]

superxoi, 2024-01-13:
This one is from another card 
[image]

superxoi, 2024-01-13:
Sometime i got this one 
[image]
It showed hekate logo but a crashing one.

superxoi, 2024-01-13:
When i enter payload section it showed none in there. But in the sd card definitely has 4 payload. Is there a possibility that my sd card reader are bad. 
[image]

superxoi, 2024-01-13:
I have just clean the sd card a bit and it showed the new one error code.
[image]
 "Hekate exception occurred (lr 84014de4) 
Undef "

CTCaer, 2024-01-13:
No need for more pics.
All of them show the issue:
[image]

APU ram solder balls or ram chips issue.
Start by reflowing apu.
You can normally see if it's aput by opening it up, pressing down and holding the apu and hit power and see if it works.

Other than that I can't help since it's a hw issue.

closed 2024-01-13 by CTCaer