CTCaer/hekate #918: Falied to init sd card

artslay, 2024-01-21:
This happens when the console is completely rebooted. It takes several attempts for the card to be detected and the console to boot
[image]

CTCaer, 2024-02-09:
You still have that issue?

artslay, 2024-02-13:
> You still have that issue?

Yes.

CTCaer, 2024-02-16:
do you see init/read errors when that happens in the micro sd info? because it looks like an issue with your sd reader.
that info is provided in hekate TUI also, when the GUI fails to load.

artslay, 2024-03-05:
> do you see init/read errors when that happens in the micro sd info? because it looks like an issue with your sd reader. that info is provided in hekate TUI also, when the GUI fails to load.

No, there were no errors.

CTCaer, 2024-03-17:
That's a hw issue or something, since you are the only one having it.
So because of that it's not easy to identify the cause and doubt it can be mitigated by software.
If you want though, you can contact me in discord. You can find my discord tag via retronx discord.

closed 2024-03-17 by CTCaer

DKingAlpha, 2024-05-17:
This is happening to me now.

I had other files load from sdcard before hekate, and hekate is load from FAT32 sdcard too, they all good until hekate runs to TUI saying failed to init sdcard. I believe this is not a hardware issue.

To solve this issue, I boot into OFW from hekate TUI, then slowly pop/insert sdcard until OFW prompts to reboot to use sdcard.

Masamune3210, 2024-05-18:
If its happening in Horizen, its a hardware issue. Nothing hekate does with the SD card would affect how OFW handles it.