CTCaer/hekate #923: [UMS] Problem with USB Mass Storage mode: transferred data is corrupted

valik-svs, 2024-02-04:
Story: I was trying to install Android/Ubuntu os to my switch OLED. After all steps from the guides I can see that bootlogo_%UPLOADED_OS%.bpm and icon_%UPLOADED_OS%_hue.bmp displayed like the file is broken/corrupted. If I  transfer these files by USB card reader - the problem is gone.

Steps to reproduce:
 
- in hekate 6.0.7  click  tools>patrition sd card>sd ums || tools>usb tools> sd card
 - transfer installer for any os
 - click "flash %UPLOADED_OS%"
 - click Home>More config> select os

expected result: booted to selected/uploaded os
actual result: data is corrupted 

<details>
  <summary>photo</summary>
   [image]
</details>

CTCaer, 2024-02-05:
1. get a screenshot from console info -> hw & fuses info
2. get a screenshot from console info -> microsd

Also you should re extract hekate from release zip here to make sure no file is corrupt

CTCaer, 2024-02-05:
You didn't say what host os you use. (And if possible, the usb chipset, e.g. intel, amedia, both, etc)
As for the rest, when I say you should re-extract official hekate, I mean it.
That includes bootloader folder AND hekate bin renamed to payload.bin

Support for piracy packs/forks can't be provided ofc.
Especially since all these, most of the time, do changes to core gpl software and violate gpl because of no source.
Can't be bothered to RE the binaries and find what was changed, aka no support.

If you restore hekate to fully official, *reboot* and you still have that problem, contact me in discord (you can find the nametag via retronx or l4s).
If not you can just use other methods inside HOS to manage files.