CTCaer/hekate #927: Cannot boot hekate - red battery icon on top left

denisapain, 2024-02-09:
If my sd card is not in, it shows the raspi logo with battery on red, i have it plugged into official charger, with sd card in red hollow battery icon on top left

how to force hekate to boot? since obviously it CAN boot since modchip displays that screen
is there a config option to ignore battery percentage
should I try disconnect and reconnect battery? It is not easy for me to open switch so that a last resort please :D

CTCaer, 2024-02-09:
Most things that are blocking in hekate can be skipped by pressing both VOL buttons. Same thing here.

Also hekate checks voltage for that and not percentage.
That means that your battery is truly discharged completely.

closed 2024-02-09 by CTCaer

denisapain, 2024-02-09:
_That means that your battery is truly discharged completely._ - does that mean I can't charge it?

denisapain, 2024-02-09:
both VOL buttons held and press power nothign change still red battery icon

denisapain, 2024-02-09:
when I did that again I went into official bootloader of nintendo??

denisapain, 2024-02-09:
**oop soryy i was able to get into hekate, ignore my other stuff!!!**