CTCaer/hekate #928: Need help . Failed when tried to dump joycon bt . switch lite

N0ahV0ss, 2024-02-21:
Hello . I have a switch lite . And I tried to install Android . But the sticks drifting in ppsspp and moonlight .
Searched , and found that I need "dump joycon bt" before install Android . But failed when pressed "dump joycon bt" .
Does a switch lite need "dump joycon bt" before installing Android ?
If switch lite don't need "dump joycon bt" . How to solve the sticks drifting ?
Thanks

CTCaer, 2024-02-22:
What's the error showing up?

N0ahV0ss, 2024-02-22:
> What's the error showing up?

[image]

just this .

First I tried and failed . Then I formatted my tf card . But failed again . Same error .

CTCaer, 2024-02-22:
Seems like you messed up with your prodinfo/cal0.
Can't help you with that, other than say that you should restore it to its original condition.

(I will update the error message with an error number to easier identify such issues in the future.)

closed 2024-02-22 by CTCaer