CTCaer/hekate #933: Suggestion - Alternating autoboot?

Clydefrosch22, 2024-03-08:
Would it be a lot of trouble to give the option to put two or more autoboot entries into the ini and have hekate alternate which one to boot after every system restart?

This would be mostly a convenience option for docked systems that are cumbersome to reach/take out of the dock (or people with physical issues, possible) to make switching back and forth between sys and emummc easier, i.e. by forcing a switch restart via something like a smart plug or clapper.
At least, you would not need to take out the switch from the dock, which currently, you'd need at least when switching from sys back to emummc. (The other way around, you can use homebrew to restart and set autoboot to sysnand)

This is the easiest way I could think of, that doesn't require hekate to support bluetooth controllers, with which you could navigate the menu blind.
Though if anyone has other suggestions, I'm all ears.

Very low priority, if it was something you'd think about, but it would be appreciated.

Leerz, 2024-03-17:
well, you can install the fastCFWswitch overlay

CTCaer, 2024-03-17:
There's 0 point to that. Just use any r2p (erista) or r2c (mariko) supported software like [studious-pancake](https://github.com/HookedBehemoth/studious-pancake).
In l4t linux you can use the built-in Reboot 2 Config app from the icon in taskbar.

closed 2024-03-17 by CTCaer

Clydefrosch22, 2024-03-17:
That only works for emummc <> sysmmc with atmosphere, right? 
I dont really like running atmo where I go online, since I might forget a mod or cheat or something else just might be set up wrong somehow, but I understand if it's nothing you want to look into.

impeeza, 2024-03-17:
You don't need a new option to select a different "autoboot"  you can disable autoboot and you are presented with a menu to you select the option you like to boot.

Clydefrosch22, 2024-03-17:
As I said in OP, to see and control that menu, a system has to be taken out of the dock, which can be a hassle when you switch a lot and the system is in a hard to reach/undock space.
I know that my setup being stupid like that is a me problem though and that this would not have been useful for most people.

impeeza, 2024-03-17:
Hekate CAN NOT WORK docked so the only way as stated by ctcaer IS TO REBOOT TO PAYLOAD, on HOS [studious-pancake](https://github.com/HookedBehemoth/studious-pancake) is the best.

Clydefrosch22, 2024-03-17:
Why are you shouting? You came in stating the obvious thing I'm obviously aware of going by my original post, ignoring the reason I asked for this (arguably niche case) feature. I was simply explaining it again. 

I could read ctcears post just fine, however, as I said, I'm reluctant to use atmosphere on sysmmc, but acknowledge that it makes switching convenient. But it's introducing a risk factor I'm not a fan of. Other people can access my switch, they could accidentally start a game holding either l or r for whatever reason, either enabling cheats/mods or ending up in the homebrew menu clicking a tile they shouldn't. 
That is in addition to a number of other things that could go wrong, like overwriting a config file when updating, suddenly reestablishing atmosphere's default of loading games with cheats and mods on. I'm sure one can do it without messing up. But I'd rather not risk it.

And I'm not sure that it completely -can't- work docked, but knowing nothing about coding, I'm sure it's a huge amount of work to implement bluetooth support (with the menu being simple enough, that could be navigated blindly without video). And I understand how that goes way beyond what is reasonable.
Videosupport, I assume, might be even more work or truly impossible

Masamune3210, 2024-03-18:
.....don't you think that if Bluetooth/Wifi/DisplayOut were possible, they would have made it into a build or have been even mentioned by now? Hekate runs super early in the boot process, almost nothing is brought up yet. There is a reason the limitations exist, I promise you its not just because he didn't want to implement it.

Clydefrosch22, 2024-03-18:
Going by this, bluetooth at least is not impossible, just a lot of work, for not enough benefit. 
https://github.com/CTCaer/hekate/issues/556#issuecomment-756465807

CTCaer, 2024-03-22:
> That only works for emummc <> sysmmc with atmosphere, right? I dont really like running atmo where I go online, since I might forget a mod or cheat or something else just might be set up wrong somehow. And while I feel like he's probably wrong, Draxzelex from the Ban-Thread on gbatemp is swearing there's a lot of evidence that bans can happen over as little as just running atmosphere. but I understand if it's nothing you want to look into.

What you suggest will have you guaranteed booting the wrong thing at times and then you will just reboot.
So, just have the autoboot option to something that can use r2p/r2c and then reboot to actual wanted thing.
Such requests, if implemented, basically create clutter for "problems" that can already be addressed more properly already.