CTCaer/hekate #934: Error After trying to migrate to new Ds

Bobyman122, 2024-03-12:
When trying to upgrade to new sd card I get this message::
What do I do?

[image]

neo305, 2024-03-12:
Did you have any themes active? Or you just need to do a fresh install of atmosphere. Delete atmosphere, bootloader folders and load a fresh copy

Bobyman122, 2024-03-12:
> Did you have any themes active? Or you just need to do a fresh install of atmosphere. Delete atmosphere, bootloader folders and load a fresh copy

In the backup I made of my sd, I did have a theme active when I backed up the sd.

urherenow, 2024-03-12:
Did you by chance have a raw based emuMMC? Both not moving that over, AND not having the correct starting sector in emummc.ini, will give you that error. It means exactly what it says. Either it can't find pkg1, or it's not a recognized version (which would mean you updated your firmware without updating both hekate and atmosphere...)

Another possibility is that your current firmware doesn't have exfat drivers and you forgot to format the new card to fat32.

Not only are there several closed issues here that cover this exact issue, that you could have looked at, but This space is generally not a help forum. You should really ask questions like this on discord retronx, or on gbatemp. A final thought: I haven't paid attention to see if the latest hekate release had a silent update or not, but I'm 99% sure there is a recent issue (that may or may not be closed) that has a fixed hekate in it... fixes a similar emummc problem. Find and grab that.

CTCaer, 2024-03-17:
Assuming you backed up your emuMMC partition and restored it, you are supposed to run the emuMMC migrate option to fix the raw partition config.

closed 2024-03-17 by CTCaer