CTCaer/hekate #936: Cancelling autoboot with - Button does not work anymore

tatundria, 2024-03-17:
Any idea why Hekate`s latest version does not allow to cancel autoboot by pressing - at boot? It is not working for me. Instead, I had to disable autoboot inside the .ini file to get into Hekate.

impeeza, 2024-03-17:
My console vulnerable using RCM Loader, I can cancel autoboot just fine.

tatundria, 2024-03-18:
I use HWfly and I cannot cancel autoboot with Hekate’s latest version…

urherenow, 2024-03-18:
Try adjusting your bootwait time up a bit...

impeeza, 2024-03-19:
by example, mine is 1 second, and I have plenty time to interrupt the autoboot:

[image]

Another thing, you will see a horizontal bar growing from left to right at the bottom of screen while you can interrupt the process.

but you can put some bigger value to test.

tatundria, 2024-03-19:
I already did that. 10 seconds. But it is not responding to - button at boot (the -button is operational)

Clydefrosch22, 2024-03-19:
just to be sure, you're pressing volume -, not select button -, right?

tatundria, 2024-03-19:
Yes. Volume - on the left joycon.

urherenow, 2024-03-19:
Vol is not on the joycon. 🤣

closed 2024-03-19 by tatundria

impeeza, 2024-03-20:
@tatundria are you joking?  are you do not using Vol- ON THE TABLET to interrupt the autoboot?