CTCaer/hekate #937: Hekate menu won't open

Htakpg, 2024-03-17:
Hey guys!!! So, in december last year I got my switch with all those thingd installed. Since college has came back, I haven't been using it for a while. It was dead, without batteries, for about a month now. Yesterday I decided to recharge it, and, as always, the hekate menu popped up, but I think I mistankenly didn't open emu/MC. I thought "Oh, no problem, I'll just restart it". Now, it simply won't open the hekate menu anymore, it just open nintendo switch in it's normal self. I really don't know what to do, I can't open any of the apps and anything. If anyone can help me I'll appreciate it so so much

impeeza, 2024-03-17:
please go to Discord, GBATemp or similar site to get support, this space is only to report BUGs.

CTCaer, 2024-03-22:
This is not a place for support on chip issues after updating hos or something.

closed 2024-03-22 by CTCaer