CTCaer/hekate #938: Auto NGC Does Not Work

harvestry-of-ghosts, 2024-03-17:
Using latest versions of everything. Hekate, Atmosphere, Nyx, sigpatches, 17.0.1, etc, etc.

Auto NGC is enabled in Hekate but I can still use game carts on emuMMC.

If I put a stratosphere.ini file in the atmosphere/config file with nogc = 1, the game card slot is properly disabled.

If I delete the stratosphere.ini file and only have Hekate's 'Auto NGC' enabled, the card slot is not properly disabled.

urherenow, 2024-03-18:
because you're using fusee in emuMMC. If you're not using fss0, you're not using hekate's config files. And you obviously did not setup nogc in Atmosphere's ini files...

harvestry-of-ghosts, 2024-03-18:
> because if you're using fusee in emuMMC, you're not using hekate's config files. You need nogc in Atmosphere's setup, in that case.
> 
> Is this statement true for you? "Automatically applies nogc patch if unburnt fuses found and a >= 4.0.0 HOS is booted"
> 
> If it's not true, then of course it won't be disabled, because there is no point.

Thank you for the reply. So to confirm that I understand you correctly... Since I already tried using Hekate fss0 instead of fusee and the slot was still not disabled, it means that there are no unburnt fuses and thus Auto NGC is working as intended. Is that correct? In which case the only option to disable the slot when a persons Switch had 17.0.1 at the time of the emuMMC creation is the stratosphere.ini setting.

closed 2024-03-18 by harvestry-of-ghosts