CTCaer/hekate #939: `hekate exception occurred (4000E854): Hang detected in LP0/Minerva > Missing LP0 (sleep) lib! Missing Minerva lib! Update bootloader folder

crossholo, 2024-03-21:
Using fw 16.0.0, AMS 1.5.2, latest Hekate release (v6.1.0) on a Hwfly/Picofly modchip (not sure which since the ebay listing had both in the description).

The switch worked fine for a day or two, until it white-screened. After a force restart, this was the error I got:
`hekate exception occurred (4000E854):
Hang detected in LP0/Minerva!`
[image]
This same error persisted even after formatting the micro-sd and setting it up again.

After swapping to another microsd, this new error came up:
[image]
`Missing LP0 (sleep) lib!
Missing Minerva lib!
Update bootloader folder!`
Not sure if the pattern in the background means anything.

impeeza, 2024-03-22:
duplicated, search on closed issues.

CTCaer, 2024-03-22:
Ram issue (at soc or ram side). I replied to you in discord also.

closed 2024-03-22 by CTCaer