CTCaer/hekate #940: Hetake exception ocurred <LR 4000A068>

RobertoJim, 2024-03-23:
Hi, 
I get this exception code when I insert hekate in the micro SD card. I can't find the source of the problem or where to find documentation about exception codes, could someone tell me how to solve it?
[image]

[image]

I have checked the micro SD slot to see if it is damaged, but everything seems to be OK. In addition, I have tried with several Sandisk micro SD, both extreme plus and plus, both 128GB.
[image]
[image]

impeeza, 2024-03-23:
Hekate works with the others cards?

urherenow, 2024-03-23:
I’m pretty sure this is the person I’ve been replying to on temp. Upon trying to launch ofw, blue screen… so it’s a hardware issue (likely from the chip install).

closed 2024-03-23 by RobertoJim