CTCaer/hekate #942: Add Support for HOS 18

nikilase, 2024-03-26:
### Issue
I am sure you might already be aware, but the new HOS 18 just released and lacks support by Hekate as it cannot boot even to stock HOS 18 by using following launch config
```
fss0=atmosphere/package3
stock=1
emummc_force_disable=1
icon=bootloader/res/icon_switch.bmp
```

For documentation sake, the error message during the failed boot to cfw/stock sysMMC using Hekate is following:
```
Unknown pkg1 version.
HOS version not supported!

Trying backup bootloader...
Found pkg1 <'20240207110330'>.

Unknown pkg1 version.
HOS version not supported.

Failed to launch HOS!

Press any key...
```

### Partial Solution
For all who have the same issue right now, you can still launch stock HOS 18 by going to `Reboot` and selecting `OFW` in the Hekate main menu.

urherenow, 2024-03-26:
This isn’t an issue. It’s expected with new HOS updates. And why would you expect using Atmosphere’s package3 to work, when Atmosphere itself hasn’t been updated to work? Don’t answer that. Just give it a rest for a week or so while the devs work on the updates.

nikilase, 2024-03-26:
I just realized this might be more of an issue with Atmosphere so I am sorry if this issue is in the wrong place and then of course it can be closed/deleted.

I am also not expecting everything to work asap (this is FOSS after all), just felt like at least having an issue so new people can see there currently is an issue with HOS18 might help.

CTCaer, 2024-03-29:
Get the latest version

closed 2024-03-29 by CTCaer