CTCaer/hekate #943: Apparently having the same problem as https://github.com/CTCaer/hekate/issues/899

mxarment, 2024-03-28:
I added my issue to that link as a comment but I notice it was closed so thats why Im opening a new one.
My response in question in that link was (read the other issue for context (https://github.com/CTCaer/hekate/issues/899)):

> Having the same problem on a friends switch OLED with the picofly,
> 
> Ive tried installing the HATS pack in the card in case some corrupted file but it always show the error "failed to init sd card"
> Also, trying that same pack on another SD card (a sandisk one) works perfectly.
> 
> I dont know what could it be but it happened from one day to another without an apparent reason, and what is more curious, I personally have a switch lite with the same chip and the exact same card bought on the same place, and it still works for me.
> 
> I dont know if it could be a manufacturing error but makes no sense because the PC can read the card and its files perfectly.
> 
> I would appreciate any ideas in case someone has some clue about what can be happening.
> 
> (Attached photos, one from my lite with the card info and the other from the oled displaying the error)
> 
> [image]
> [image]

urherenow, 2024-03-28:
If you're having the issue with picofly, then you can't blame your problem on hekate. Use your head.

AnalogMan151, 2024-03-28:
... I said I'm having problems in both? So even if I get passed the Picofly issue I'm still having init issues in Hekate, which is why I added my comment here.

urherenow, 2024-03-28:
If the chip itself can't initialize the card, it can't load hekate from it. Figure that out first. Swapping cards back and forth, thinking that once nyx is loaded it will be able to initialize a card that the chip can't, is not a reasonable expectation.

AnalogMan151, 2024-03-28:
If the card is formatted properly and works in HOS then the reasonable expectation is that Hekate can init the card. It can't, so that's my focus right now, thank you anyway though.

mxarment, 2024-03-28:
> If you're having the issue with picofly, then you can't blame your problem on hekate. Use your head. You might need to have this addressed at https://github.com/Ansem-SoD/Picofly. But not knowing what firmware you have on your picofly, I'd grab the latest from there and flash it first, before creating an issue.

Yeah you right, I should make the post in the Picofly's github but because I saw that exact issue here from this other guy https://github.com/CTCaer/hekate/issues/899 I just went reopening a new one here, is not a big deal

urherenow, 2024-03-28:
> If the card is formatted properly and works in HOS then the reasonable expectation is that Hekate can init the card. It can't, so that's my focus right now, thank you anyway though.

The OP shows that same card working on a Switch Light, so hekate clearly CAN. The problem is with your chip. Is yours also an OLED? Many people installing picofly on their own, have found that OLED requires fiddling with the resistor values in up to 3 different places at times. Hekate is read first and foremost, from the card. If the chip itself can’t read the card, you need to fix that first.

CTCaer, 2024-03-29:
I can't just magically fix issues caused by hungry modchips on the 3v3 main regulator like that.
And more importantly, modchip support is not provided here.

closed 2024-03-29 by CTCaer

AnalogMan151, 2024-03-30:
Alright, I'm commenting my findings here to help those affected. I regret bringing up the modchip as it seems to be stealing all the focus despite the original issue being Hekate not mounting the SD card and the modchip behavior simply being mentioned for additional info. While I didn't find the cause of Hekate's inability to init the SD card I was able to replicate the issue and find a work around.

Purchasing a brand new Samsung EVO card (this one was 512GB) and inserting it into a Nintendo Switch that does not have the exFAT update, declining the update prompt to use the card, and then going to System Settings to format the new Samsung EVO card to FAT32 results in an SD that the Switch and a computer can read / write to just fine but that won't init in Hekate.

If you then quick format the same card using GUIFormat (it defaulted to 32K allocation and I left it) then the issue goes away and Hekate can successfully init the card. 

I'm not sure if it's something to do with how the Switch formats the card, how Samsung sets up the initial primary exFAT partition it comes pre-setup with or some combo of the two, but if you use the Switch itself to reformat to FAT32 then it creates an issue with Hekate. Quick formatting with a PC, despite not changing any partition layout or info, does seem to resolve the issue. 

This was replicated with a new 512GB card I bought thinking my old one may have been faulty. Once I got the new 512GB card working in Hekate, I backed up my previous card, performed the quick format, restored the backed up files and now the card works fine in Hekate. 

As far as I am concerned I am now satisfied and hopefully @mxarment and those affected by #899 may benefit. If any of you initially formatted your Samsung cards using the Switch itself back when you first purchased the cards, maybe try backing up and quick formatting the cards with GUIFormat on a PC.

mxarment, 2024-03-30:
> Alright, I'm commenting my findings here to help those affected. I regret bringing up the modchip as it seems to be stealing all the focus despite the original issue being Hekate not mounting the SD [...]

Thank you for sharing, Ill try that asap ♥️