CTCaer/hekate #944: (v6.1.0) Unknown pkg1 version, can't boot from any payloads or stock

IronFoxGaming, 2024-03-29:
I recently updated my switch while trying to get to the YouTube app after a long day at work, and now I can't get my switch to boot into any payloads at all, and the stock option doesn't work either. I thought at first that it had to do with me using v6.0.7 before, but after updating to the latest version I still got this error. I searched around, and did my best to update Atmosphere and Fusee too, but I'm still getting the same error after that. I don't have a way to restore my switch to how it was before the update, because I have had no SD cards that'd be able to hold a whole system recovery for it to dump into. I found an issue on here before, where the original person just didn't know how to update or read instructions, and they were told to go to the discord server for hekate to ask for help, but I can't find anything about the server anywhere online. I just keep getting the switch discord rich presence stuff. Anyone know what I should do?

Tldr: I am at the latest version of all cfws and I keep getting the unknown pkg1 error for every payload boot type no matter what I do.

urherenow, 2024-03-29:
Issue #942 

FW18 was just released. You need to wait for both hekate and Atmosphere to get updated.

CTCaer, 2024-03-29:
Get the latest version

closed 2024-03-29 by CTCaer