CTCaer/hekate #945: "Failed to apply emummc" under Hekate 6.1.1

muxi1, 2024-03-29:
Hi,
There seems to be a problem with the new Hekate version 6.1.1. When booting the emuMMC, the error message "Failed to apply emummc!" "Failed to launch HOS!" is displayed.

gnxswitch, 2024-03-29:
Same problem here. Came to post this.

yoho2chn, 2024-03-29:
I got blue screen when Launch stock, but reboot to OFW18.0 boots up fine.

Kisora1, 2024-03-29:
I also have this same issue, I've never had it before

yoho2chn, 2024-03-29:
did you guys also update to ATM 1.7.0 pre-release as well?

Kisora1, 2024-03-29:
> did you guys also update to ATM 1.7.0 pre-release as well?

I did

GordiNoki, 2024-03-29:
Yes, and applied sigpatches from https://gbatemp.net/threads/sigpatches-for-atmosphere-hekate-fss0-fusee-package3.571543/post-10388870

muxi1, 2024-03-29:
CTCaer has fixed the problem by updating the version. You simply have to download the release again!

closed 2024-03-29 by muxi1

Kisora1, 2024-03-29:
Where is the update? I don't see it

Kisora1, 2024-03-29:
Nvm I fixed it by using AIO Switch Updater

DietDrSlurm, 2024-03-29:
#947 If you are having issues even after updating to the newest version, there may still be a bug and rebooting from a fresh RCM instance instead of using reboot to payload could solve your issue.

rursache, 2024-04-01:
@DietDrSlurm 

having the exact same issue on a chipped mariko. i need to run the `hekate_ctcaer_x.x.x.bin` from payloads in order to succesfully boot the emuMMC. i have the latest release downloaded 1h ago.

otherwise i get `failed to apply emummc`. 

i updated hekate since HOS 14.x, never had this issue before. im doing it correctly im sure!

impeeza, 2024-10-01:
> i have an older version of hekate (v6.0.4) and after like a year not opening the switch, the screen when i want to open the modded launcher just shows failed to init emuMMC and failed to launch HOS. i have 0 idea about switch modding, as i got mine modded at a shop. do you guys have any sort of fix that i can do?

Buddy, write on an CLOSED issue asking for help, is no polite, please seek for help on Discord or GBATemp.