CTCaer/hekate #947: Inconsistent emmMC behavior

DietDrSlurm, 2024-03-29:
Under the most recent as of an hour ago Hekate v6.1.1 & Nyx v1.6.1 that aimed to fix the "failed to apply emummc" error, I am able to fully boot from a fresh RCM, but am getting persistent "failed to apply emummc" when I try to reboot by using the rebooting to payload from Atmosphere and relaunching from the Hekate/Nyx dashboard.  Reloading from the Hekate/Nyx dashboard doesn't fix the issue, but rebooting to a new RCM instance either through Hekate or a hard reset will allow the switch to boot successfully boot again.

rursache, 2024-04-01:
having the exact same issue on a chipped mariko. i need to run the `hekate_ctcaer_x.x.x.bin` from payloads in order to succesfully boot the emuMMC. otherwise i get `failed to apply emummc`. i updated hekate since HOS 14.x, never had this issue before. im doing it correctly

DietDrSlurm, 2024-04-04:
Confirmed to still be an issue.  I replaced bootloader folder and Hekate bin with current 6.1.1/1.6.1 as of today and issue persists.  Erista switch on 18.0 Firmware. with Atmosphere 1.7.0-pre.

SellPL, 2024-04-20:
Can also confirm this behavior on latest hekate! OP's method of booting saved me from immediate heart attack haha

CTCaer, 2024-05-10:
Just extract the latest release zip in sd to make sure update.bin which is used in the case of erista is also replaced.
Just changing the injected payload is not enough when you do r2p.

closed 2024-05-10 by CTCaer