CTCaer/hekate #948: UMS SD card isn't working

x676278, 2024-04-01:
When I select tools-USBtools-SD card, the other machines show waiting to connect, my machine shows USB activated, no prompts after connecting using the cable, but the switch does not show up in my PC drive list at all, there is no way for me to securely pop it up on the PC side, but my Another switch works fine in the same operating environment, I don't know what other information I need to provide, please let me know if you can, this problem has been bothering me for a long time and I haven't been able to find a solution, please help me!
[image]

impeeza, 2024-04-01:
can you please try with the official release from https://github.com/CTCaer/hekate/releases and repeat the test?

please also send a screenshot of the microSD Card Info Page ( Console info -> storage & battery info, microSD)

thanks.

x676278, 2024-04-01:
> can you please try with the official release from https://github.com/CTCaer/hekate/releases and repeat the test?
> 
> please also send a screenshot of the microSD Card Info Page ( Console info -> storage & battery info, microSD)
> 
> thanks.

I have updated to the latest version, but the problem is still there, here is my SD card information, please help me analyze it, thank you very much for your reply!
[image]
[image]
[image]

impeeza, 2024-04-02:
Well ExFAT can be prone to corruption you should use FAT32 on the SD Card.

But in this case can you please test the USB connection by using the gamepad emulation on the same page is ums. With that were can check the port, cable and connector.  You should see a USB gamepad connected to your PC

x676278, 2024-04-02:
> Well ExFAT can be prone to corruption you should use FAT32 on the SD Card.
> 
> But in this case can you please test the USB connection by using the gamepad emulation on the same page is ums. With that were can check the port, cable and connector. You should see a USB gamepad connected to your PC

it's worked，but UMS SD card still fail
[image]

x676278, 2024-04-02:
[image]
[image]

impeeza, 2024-04-02:
With this we checked the PC USB Port, The Cable, The Console USB Connector, The USB Stack on NYX.

here any developer could give us a hint.

CTCaer, 2024-05-10:
Play with `bpmpclock=` in `nyx.ini` and it will solve your issue (read the readme). If not reopen the issue.
Also have in mind that no support is generally given for modded versions of hekate/Nyx.

closed 2024-05-10 by CTCaer