CTCaer/hekate #949: HOS panic occurred!

OblivionDarkness, 2024-04-01:
color: ######, code: 06

have no idea what this error is top do with any help

Duffy0079, 2024-04-02:
hey  man, just made a github account to reply to this. I literally had the same issue with my switch yesterday after updating the os to version 18.0.0. I'm assuming you did what I did and just went to the nyx git hub page and downloaded the update and tried to copy files. what ended up being the fix for me and my friend was to go to the atmosphere git hub https://github.com/Atmosphere-NX/Atmosphere/releases. download the 1.7.0 pre release and copy/replace those files with the ones on your sd card and your switch should boot into cfw like normal. truly hope this helps

josh-vin, 2024-04-09:
Wow! @Duffy0079 you are amazing. That is exactly what happened for me. I updated my files to the prerelease ones and its working great again. I thought I bricked it. 

Just for context for anyone else. I also updated to 18.0.0 and updated the firmware with Daybreak and then I rebooted and couldn't get my switch to load again.

Thanks a million for posting here!

CTCaer, 2024-05-10:
You always need to update both hekate and atmosphere on hos updates.
A blue or yellow screen or error code, basically just tells you that.

closed 2024-05-10 by CTCaer

Lil-Mouse-69, 2024-06-22:
> hey man, just made a github account to reply to this. I literally had the same issue with my switch yesterday after updating the os to version 18.0.0. I'm assuming you did what I did and just went to the hekate git hub page and downloaded the update and tried to copy files. what ended up being the fix for me and my friend was to go to the atmosphere git hub https://github.com/Atmosphere-NX/Atmosphere/releases. download the 1.7.0 pre release and copy/replace those files with the ones on your sd card and your switch should boot into cfw like normal. truly hope this helps.

TY so much this fixed my problem in an instant ty