CTCaer/hekate #951: nyx exception occured (LR 81065db4) undef

MarcinhoLetsCode, 2024-04-05:
Sometimes hekate freezes too

Almasora5, 2024-05-07:
I have the same problem and I cannot do anything in hekate, has anyone solved it?

giannicolicchia, 2024-05-09:
Any news on this?

MarcinhoLetsCode, 2024-05-09:
if you guys are trying to back up sysnand and create an emunand, i found a way of doing that:
1-to backup sysnand i used an earlier version of hekate (hekate_ctcaer_4.10.1), without GUI,
2-to create a emunand i had to use (hekate_ctcaer_5.1.0_Nyx_0.8.3), gave me error on first try, but worked when i tryied again
3-Updated hekate and everything normally

*if boot entries are empty:
1-( https://github.com/CTCaer/hekate/issues/456 ) 
2-( https://rentry.org/EristaEmuNAND )

*if some error about "Unknown pkg1 version" occurs:
1-( https://rentry.org/EristaEmuNAND )

**Be Careful Following**
**Try At Your Own Risk**

CTCaer, 2024-05-10:
Piracy is not supported

closed 2024-05-10 by CTCaer

Maario01, 2024-05-24:
Same here... No solution?

impeeza, 2024-05-25:
> Same here... No solution?

Piracy is not supported

> Piracy is not supported

MarcinhoLetsCode, 2024-05-25:
Sadly The only solution i found was using an earlier version of the program :/

MarcinhoLetsCode, 2024-05-25:
Help us to solve this error