CTCaer/hekate #954: Gamepad HID emulation: spontaneous inputs, mostly stick jumps to extreme and back

v1993, 2024-04-10:
I've been using HID Emulation on Linux and have noticed that spontaneous inputs are a bit of an issue - and no, I'm not talking about joycon drift. Most often, these manifest as left stick jumping to the bottom left and then nearly instantly back - but this is enough to register as an input in games. Example log from evtest, Switch remained untouched for the duration of the test:

```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x11ec product 0xa7e1 version 0x110
Input device name: "Nyx USB Nyx Joy-Con"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 288 (BTN_TRIGGER)
    Event code 289 (BTN_THUMB)
    Event code 290 (BTN_THUMB2)
    Event code 291 (BTN_TOP)
    Event code 292 (BTN_TOP2)
    Event code 293 (BTN_PINKIE)
    Event code 294 (BTN_BASE)
    Event code 295 (BTN_BASE2)
    Event code 296 (BTN_BASE3)
    Event code 297 (BTN_BASE4)
    Event code 298 (BTN_BASE5)
    Event code 299 (BTN_BASE6)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 1 (ABS_Y)
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 2 (ABS_Z)
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 5 (ABS_RZ)
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1712710841.125522, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1712710841.125522, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712710841.125522, -------------- SYN_REPORT ------------
Event: time 1712710841.170565, type 3 (EV_ABS), code 0 (ABS_X), value 127
Event: time 1712710841.170565, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712710841.170565, -------------- SYN_REPORT ------------
Event: time 1712710886.150589, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1712710886.150589, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712710886.150589, -------------- SYN_REPORT ------------
Event: time 1712710886.180643, type 3 (EV_ABS), code 0 (ABS_X), value 127
Event: time 1712710886.180643, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712710886.180643, -------------- SYN_REPORT ------------
Event: time 1712710888.227701, type 3 (EV_ABS), code 2 (ABS_Z), value 254
Event: time 1712710888.227701, type 3 (EV_ABS), code 5 (ABS_RZ), value 254
Event: time 1712710888.227701, -------------- SYN_REPORT ------------
Event: time 1712710888.258645, type 3 (EV_ABS), code 2 (ABS_Z), value 127
Event: time 1712710888.258645, type 3 (EV_ABS), code 5 (ABS_RZ), value 127
Event: time 1712710888.258645, -------------- SYN_REPORT ------------
Event: time 1712710920.118681, type 3 (EV_ABS), code 2 (ABS_Z), value 0
Event: time 1712710920.118681, type 3 (EV_ABS), code 5 (ABS_RZ), value 254
Event: time 1712710920.118681, -------------- SYN_REPORT ------------
Event: time 1712710920.148643, type 3 (EV_ABS), code 2 (ABS_Z), value 127
Event: time 1712710920.148643, type 3 (EV_ABS), code 5 (ABS_RZ), value 127
Event: time 1712710920.148643, -------------- SYN_REPORT ------------
Event: time 1712710955.163688, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1712710955.163688, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712710955.163688, -------------- SYN_REPORT ------------
Event: time 1712710955.193696, type 3 (EV_ABS), code 0 (ABS_X), value 127
Event: time 1712710955.193696, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712710955.193696, -------------- SYN_REPORT ------------
Event: time 1712710960.396696, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1712710960.396696, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712710960.396696, -------------- SYN_REPORT ------------
Event: time 1712710960.426678, type 3 (EV_ABS), code 0 (ABS_X), value 127
Event: time 1712710960.426678, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712710960.426678, -------------- SYN_REPORT ------------
Event: time 1712710978.657712, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1712710978.657712, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712710978.657712, -------------- SYN_REPORT ------------
Event: time 1712710978.688706, type 3 (EV_ABS), code 0 (ABS_X), value 127
Event: time 1712710978.688706, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712710978.688706, -------------- SYN_REPORT ------------
Event: time 1712711000.444783, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1712711000.444783, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712711000.444783, -------------- SYN_REPORT ------------
Event: time 1712711000.489769, type 3 (EV_ABS), code 0 (ABS_X), value 127
Event: time 1712711000.489769, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712711000.489769, -------------- SYN_REPORT ------------
Event: time 1712711131.203931, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712711131.203931, -------------- SYN_REPORT ------------
Event: time 1712711131.233932, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712711131.233932, -------------- SYN_REPORT ------------
Event: time 1712711145.542977, type 3 (EV_ABS), code 2 (ABS_Z), value 0
Event: time 1712711145.542977, type 3 (EV_ABS), code 5 (ABS_RZ), value 254
Event: time 1712711145.542977, -------------- SYN_REPORT ------------
Event: time 1712711145.587935, type 3 (EV_ABS), code 2 (ABS_Z), value 127
Event: time 1712711145.587935, type 3 (EV_ABS), code 5 (ABS_RZ), value 127
Event: time 1712711145.587935, -------------- SYN_REPORT ------------
Event: time 1712711223.312026, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1712711223.312026, type 3 (EV_ABS), code 1 (ABS_Y), value 254
Event: time 1712711223.312026, -------------- SYN_REPORT ------------
Event: time 1712711223.343037, type 3 (EV_ABS), code 0 (ABS_X), value 127
Event: time 1712711223.343037, type 3 (EV_ABS), code 1 (ABS_Y), value 127
Event: time 1712711223.343037, -------------- SYN_REPORT ------------
```

Considerably less often, what seems to be spontaneous button presses also happen - while I could theoretically blame my clumsy hands for *some* of that, "+" button being pressed when my fingers are nowhere near it or inputs happening while I'm away from the PC does not seem exactly likely.

I've verified that none of those issues happen in other situations, such as playing on switch or using joycons over bluetooth with PC - or, indeed, using hekate, where stick controlled pointer does not appear out of nowhere now and then if left alone. Removing one of joycons prevents faulty (and non-faulty) events from being generated. `dmesg` does not seem to reveal anything interesting.

CTCaer, 2024-05-10:
What switch SKU you are using that on?

v1993, 2024-05-10:
Official joycons. SKU `83 - Icosa (Erista) - Retail`.

CTCaer, 2024-05-10:
Ok, I will investigate when possible.
I will post a nyx.bin here for you to test when I have something.

CTCaer, 2024-05-19:
@v1993 try this:
[nyx.zip](https://github.com/CTCaer/hekate/files/15369728/nyx.zip)

v1993, 2024-05-19:
The issue seems to be fixed with the version you've attached, thanks a lot! I didn't have a chance to test with a game, but evtest remains clean when controller is left laying untouched and shows correct inputs when these are made.

CTCaer, 2024-06-11:
Fixed on 1.6.2 Nyx.

closed 2024-06-11 by CTCaer

CTCaer, 2025-11-15:
@v1993 you should give latest Nyx 1.8.0 a try btw. You should be able to see even better behavior.
Basically, there were some major changes with how USB HID works internally. And most importantly it now supports input report throttling.