CTCaer/hekate #955: I think I mess up

LastCinder, 2024-04-14:
I've been dealing with a problem for a few days, which isn't an error within the code or anything like that, it's something more. I made the mistake of activating an option that causes the console to automatically start in the wrong payload, skipping all of the hekate part, and now I want to undo it... I hope you can help me, and I'll be reading. 
I'm attaching the video of someone who did it: https://www.youtube.com/watch?v=4De5IDa-Ovc

urherenow, 2024-04-14:
You should have just read BEFORE using hekate, much less, creating an issue here. Edit your hekate_ipl.ini, or read what button overrides the auto-start. Probably vol- or something…

impeeza, 2024-04-14:
> I've been dealing with a problem for a few days, which isn't an error within the code or anything like that, it's something more. I made the mistake of activating an option that causes the console [...]

This is not a support site, please go to GBATemp.net or discord. You can cancel the auto boot of Hekate pressing VOL DOWN BUTTON during boot

CTCaer, 2024-05-10:
VOL- can always be used, no matter what boot wait is set to when auto boot is enabled.
If wait is 0, you just need to press and hold it during hekate init.

closed 2024-05-10 by CTCaer