CTCaer/hekate #957: emuMMC wont start - error 2002-4153 program: 0100000000000031

santiagoahumada13, 2024-04-24:
when i try to enter mi emummc partitios i get this error after the nintendo logo, ive tried updating everything, even overlay modules, deleting cofig folder and i still cant access my system, help please 
[image]

impeeza, 2024-04-24:
Backup and delete SD:/atmosphere/contents folder.

Normally is an outdated sysmodule.

please next time ask for help on GBATemp or Discord, https://gbatemp.net/threads/homebrew-crashes-indicate-0100000000000031-as-culprit.632198/

CTCaer, 2024-05-10:
Or corrupted data or mismatched. Maintenance mode might also fix it.
But for sure here's the wrong place to get support for such things.

closed 2024-05-10 by CTCaer