CTCaer/hekate #960: Cannot mount BOOT0/BOOT1 via the Hekate USB Tools

sthetix, 2024-05-08:
I am using the Hekate version 6.1.1 and when trying to mount the BOOT0 and BOOT1 via the USB Tools , this error comes up.

[image]

How to fix this issue?

CTCaer, 2024-05-10:
Your OS is trying to access more than it should.
It's not a fatal error from hekate side (reason why I have the text as warm), so it falls on the OS.
If that happens the moment you mount it, just use a proper OS.
(Win 7, win 10, Linux and whatever random mac os I use to test UMS do not do that.)
If you wrote some random MBR or GPT in boot0/1 for whatever reason then remove it or fix it to the proper size.

I'll leave it open, in case you have more info to add about what eMMC you are trying to mount and OS.

sthetix, 2024-05-10:
You were right.I tested it on Windows 10, and it did not display the error
message.

CTCaer, 2024-05-10:
Since that's not normal to happen, you should check if you have any mass storage or usb driver that replaced official ones.
Or any hidden app that parses drivers. Because it can be the automounter's problem.

If not then that OS is garbage because when it is actually informed about storage size, it must never try to read more.
That info is sent during the inquiry command. You can't mount a driver without it's size populated or not sent or requested.
So, there's no point to try and read/write out of bounds and also it can hang/halt hw that does not check if it's above.
For example if storage on a ums device is eMMC or SD and you try to read out of bounds, it tears down the whole thing.
You need to power cycle it after that. It's not a hot recoverable error

closed 2024-05-10 by CTCaer