CTCaer/hekate #961: [Feature Request] Launch confirmation prompts

zamastyle, 2024-05-08:
Some launch configurations are riskier to use than others. As a modder, it would be a huge benefit to be able to add a custom launch confirmation prompt to the launch configuration block. 

When a launch configuration is selected to launch
- if there a prompt key in the configuration block
---- a confirmation modal pops up with the prompt string above a "Launch" and "Cancel" button
---- if launch, launch as usual
---- if cancel, return to launch selection menu
- if there is no prompt key in the config
---- launch immediately

Being able to inform a user of the limitations or potential issues with what they are about to launch could really help prevent mistakes with new users and risky clicks.

zamastyle, 2024-05-10:
Ok. I assume this would extend to rejected a PR containing this feature if I implemented it myself?

CTCaer, 2024-05-10:
If you are gonna implement it anyway for custom builds, you may as well try to PR it.
But probably it's gonna be rejected. The reasons you gave are not enough for this, which I consider bloat.
Like what is a risky config even? There are no such things except if you are distributing malicious things.