CTCaer/hekate #964: Add dump/restore Battery Charger IC Regs.

KanzlerDev, 2024-05-22:
There's a weird bug with battery charger IC. Factory CHARGE (not INPUT) Voltage Limit TI bq24193 is 4320mV. In some cases the Reg can reduce to 4192mV because of bugs and we are loosing about 15% full battery capacity. First time i noticed this issue after using Battery Desync. I've decided my battery degraded and bought new Switch. My second Lite got this problem after TegraExplorer Wipe (HOS 17). I could only investigate the problem linked with reseting some Regs via i2c (address 7000c000 at 0x006b) during some bugs in native homebrew, payloads, switchroot Android and/or L4T. After reducing the Reg, Switch battery stop charging at 4.19V (3100mA +/-). I installed this battery in another Switch (without this issue) - the battery charged till 4.33V (3620mA +/-). I tried charging my Lite in HOS during several weeks, used Desync - nothing helped. I found some guys with the same issue. No one knows how to fix it. Curent Hekate version can only dump Battery fuel gauge regs (as i understood from max17042 via i2c at 0x0036) but restoring the dump will not help. This is the reason i create the request.

CTCaer, 2024-05-22:
It doesn't work at all like that. Like at all.
There's no weird bug and the factory (reset) charge voltage for the IC is not 4320mV. It's 4208mV.
And the "investigation" you did is false. No homebrew, payload or L4T can cause that, except if you went out of your way to mess with these.
For Android, I don't know if it took my changes for actually parsing the used charge voltage limit.

In any case, **all these are irrelevant**. **Only HOS decides the voltage limit**.
That means that even if you restore or change bq24193 regs or if you messed up with the above, the limit will change again the moment you boot HOS.

So the only way to mess that up is:
1. By writing bs on the charge cycle register in max17042 (since you mentioned doing random things with i2c)
 If that's done and you boot it gets saved in HOS savedata and gets reset
2. Messing up HOS savedata on any way
3. Restoring messed up HOS calibration data into the HOS savedata

So the "bugs" are basically a user doing random things without understanding what these can cause.

Again, HOS decides the voltage limit. And it decides it based on its savedata charge cycles value.
And it selects between these voltages: 4320, 4304, 4288, 4272, 4256, 4240, 4224, 4192.

You will need to fix your savedata if HOS decided wrong for any of the above reasons.

If not, I won't ever provide any help on resetting that, since it's a fire hazard.
3.8V batteries are not supposed to be charged at 4.3V after a certain number of charge cycles defined by their manufacturer.

PS. If you reply with a wall of text without structure like that it won't get answered next time.

closed 2024-05-22 by CTCaer

CTCaer, 2024-05-23:
And something important to note; stop using system wipers for HOS for no reason.
At this time of writing, every tool does it wrong and misses very important settings that need to be repopulated.
HOS can't do that on its own, factory uses special apps for that.

hukaka22, 2024-08-13:
Oh.no
I have the same problem.I don’t have backup.
How to solve it？