CTCaer/hekate #966: Linux Flashing | Over filesize limit for FAT32

SaltMaster420, 2024-06-01:
I'm still working through trying to flash my OG Switch with Ubuntu Noble Numbat.
I notice the download mentioned has 2 main data files l4t.00 and l4t.01, the l4t.00 is 4.3GB and trying to transfer to the FAT32 partition in the guide fails due to the FAT32 limitation.
exFat let me install linux but I am stuck at a booting issue.

I'm presently in the process of using the documentation to merge the files back to a single image, then resplit to a custom 3800M or so size to re-test, will update on my findings.

SaltMaster420, 2024-06-01:
This may be due to a faulty SD card, will close if so.

SaltMaster420, 2024-06-01:
Worked with different SD card.

closed 2024-06-01 by SaltMaster420

Monstermash28425R1, 2024-06-10:
hello sir, where did u find the ubuntu image? i can't find it on the internet

impeeza, 2024-06-10:
> hello sir, where did u find the ubuntu image? i can't find it on the internet

Buddy, this is not a support forum, you can go to GBATemp.net or discord and you will get help there quickly.

On a search using google for Nintendo Switch Ubuntu, THE VERY FIRST SITE IS:

https://wiki.switchroot.org/wiki/linux/l4t-ubuntu-bionic-installation-guide