CTCaer/hekate #967: Failed to init or mount SD!

CHNWangXi, 2024-06-02:
I have a strange Switch.It can only be used with a specific version of hekate below 5.70.If the version of Hekate over 5.7.0,it may throw error 'Failed to init or mount SD!'.I suspect this is caused by a bug in hekate.Can you help me troubleshoot the problem?

My switch version is the oldest one.

CTCaer, 2024-06-02:
If it was a bug with hekate, you wouldn't be the only one having that issue.

The only relevant change in 5.7.0 is that I removed the million sd mounts and streamlined it to only one.
That means that if that sd mount (completely) fails 5 times, it will show the error.
Previously it would try way too many times (up to 1-2min, can't remember).
So you should probably fix your sd reader.

How you inject hekate btw?

CHNWangXi, 2024-06-02:
I have tried the following operations:
1. I tried replacing multiple brands of sd cards, all ineffective;
2. I tried changing the injection method, tried using TegraRcmGui, PicoFly chip, tx chip, all invalid;
3.finally I tried to replace the version of Hekate, and found that it works below version 5.7, but not above that;

CTCaer, 2024-06-02:
If you have a sd reader connector issue, no amount of sd card changing can fix it.
In actuality all that inserting/ejecting makes it worse.

Anyway, until you fix that, you can do this change:
<https://github.com/CTCaer/hekate/blob/master/bdk/storage/sd.c#L106>
Change that line to
```c
		sd_mode = SD_UHS_SDR104;
		type = SDHCI_TIMING_UHS_SDR104;
		break;
```
This will indefinitely try to mount sd card.

I already try to init sd card in 1bit mode for broken connectors that have DAT0 intact.
But if you have issues on DAT0/CMD/CLK, there's not gonna be an infinite retry like that in an official release.
Basically, have in mind that HW issues are out of scope for hekate.

closed 2024-06-02 by CTCaer

CHNWangXi, 2024-06-02:
I think it's not a hw problem，because everything is ok in ofw mode.And i checked all the pin of the sd connector is ok.If necessary i can record a video demonstrate this issue.

CTCaer, 2024-06-03:
It seems that you are not reading what I'm saying properly.
Plus, you are the only one that has that issue and you still think that it's not a hw issue.

So again, you are saying that 5.7.0 "doesn't work" but 5.6.5 does (and for the older/newer respectively).
The only relevant change between the two is removal of the "paranoid" sd init. From 30+ full reinits to 5.
Also you think that everything is ok in OFW but you have no way of actually validating that.
Only speed changes and sd disabled or not. Because HOS also tries to reinit sd several times and on different speeds/modes.

As I said, try to compile hekate with the change I told you and see what happens. As for the sd reader, I'm talking about the actual connector pins, not the sd reader pins. But both need to be checked. And the connector pins need to be checked on the mobo and the sd reader side.
And since you mingled with the motherboard, you should also check that you didn't knock off the sd related resistors/capacitors or calibration resistors.

CHNWangXi, 2024-06-07:
Through my inspections over the past few days, I finally discovered the cause of the problem:
1. this machine has some screen problems, although it does not affect the display, but using Hekate version 5.7 or above will cause the SD card initialization failure;
2. before I said that the version of Hekate below 5.7 is ok, then I carefully compared the difference between 5.7 and 5.6, and finally determined that the display_init() function in the ipl_main() function location change, in this machine will lead to the problem of SD card initialization failure, I moved this function to the same location as 5.6, this failure disappeared;
3. Finally I replaced the screen on this machine and used the original Hekate, which also does not cause the SD card initialization failure problem;

The above is the problem I found in the past few days of research on this machine, I hope guru can take the time to help look at why the screen has a little  problem will lead to display SD card initialization failure, thank you!

Translated with DeepL.com (free version)

CTCaer, 2024-06-07:
I can only assume some short near the sd traces from DSI (display) traces which affects sd card init (for example the comp resistor).
So by initializing sd card first it would mitigate/bypass the issue.
And if display is initialized first then the short starts affecting sd card.

Thing is that you never entered console info -> sd card in TUI to see the error messages from init.
So you can't know which pad was problematic.

CHNWangXi, 2024-06-07:
If I use the original Hekate without the TUI（the orininal can't into the TUI）, the SD info shows 5 init errors, if I use Hekate compiled after adjusting the position of display_init(), the init error on the SD card info is 0. 

So I think initializing the screen first, and then it just so happens that there's something wrong with the screen on this machine that's causing this weird problem.

Thank you very much for your answer.

CTCaer, 2024-06-07:
Deleting bootloader/sys/nyx.bin allows TUI to show up btw (Nyx is basically the GUI).

CHNWangXi, 2024-06-07:
[image]
This picture shows running the original Hekate 6.1.1

[image]
And this picture shows running my own compilation of Hekate 6.1.1