CTCaer/hekate #969: HOS panic occurred!

PreHeX, 2024-06-11:
Trying to launch emuMMC from [hekate v6.2.0 & Nyx v1.6.2] with [Atmosphère 1.7.1] on firmware 18.0.0. on an "unpatched Nintendo Switch v1"
I launch it and get following Error Message:

HOS panic occured!

Color: ####, Code: 01
Press any key...

the Color of the "####" is blue.

What do i do?

Raph9213, 2024-06-11:
Try redownloading the latest hekate version, v6.2.0 had a bug on the first release where it couldn't boot before on all v1 switch

CTCaer, 2024-06-11:
Yep I broke booting on all eristas with a typo mistake. The redownload should fix that.
I will keep the issue open for now.

hanstoribio, 2024-06-11:
I downloaded the new version and I still have the blue screen problem

BlytheScythe, 2024-06-11:
Same here, I'm still getting the blue screen and I have an Erista. I've tried booting into Atmosphere via fusee.bin and my emuMMC works, it's just that loading Atmosphere's package3 via Hekate's FSS0 doesn't work for some reason. If you need anything else that I can check, let me know.

muxi1, 2024-06-11:
Shut down your consoles completely and reboot. Then the error will be fixed.

mayt04, 2024-06-11:
I will try,thank you!

hanstoribio, 2024-06-11:
I confirm, it doesn't work

PreHeX, 2024-06-11:
I formatted the SD Card just to be sure and installed Hekate,Atmosphere and the Sigpatches again (all from the latest release) and made a fresh emuMMC and tryed to launch...

same error Unfortunately

hanstoribio, 2024-06-11:
If you enter from the hekate menu and choose sysnand or emmu it does not work, but if you inject the direct fuuse.bin it does work

CTCaer, 2024-06-11:
Well either github or the browser seems to be caching it.
Someone else even reported BlobNotFound github error or the link not working at all.
The updated zip should have MD5:  20cb6c523a61512651e7070a343b8f4d 

Also r2p loads the payload on boot, so a power off is mandatory.
Finally while extracting it, the option that replaces files and merges folders should be selected.

muxi1, 2024-06-11:
I've heard from other users who are still having issues with 512GB, 1TB and 1.5TB capacities, especially under the OLED Switch.

mayt04, 2024-06-11:
The zip i downoalded has this MD5 Hash: 20CB6C523A61512651E7070A343B8F4D
Perhaps I need to delete my bootloader folder from my microSD and then put the one from the zip?

hanstoribio, 2024-06-11:
I installed everything from scratch and the blue screen still appears (I'm using the new file)

CTCaer, 2024-06-11:
That's the correct zip hash. How you inject it? PC or loader?

mayt04, 2024-06-11:
> That's the correct zip hash. How you inject it? PC or loader?

I used the Hekate's MTP responder,maybe I should turn on the console,put the SD on the PC and then put the folder?

PreHeX, 2024-06-11:
the latest .zip indeed has the MD5 Hash: 20cb6c523a61512651e7070a343b8f4d and i replaced it in the SD card i even tryed different ways over USB and Sd card removing still same error

muxi1, 2024-06-11:
It worked 1-2 times, now I get a blue screen again. The payload is loaded via a dongle.

mayt04, 2024-06-11:
Also I use TegraRCMGui with the latest hekate.bin

BlytheScythe, 2024-06-11:
I've downloaded and extracted to SD card the correct .zip file.
I personally use Rekado to push my payloads.

I still run into the same error.

I've tried clearing the system cache and any pending update, didn't fix it.
Powered off, booted into RCM and pushed the latest Hekate payload and when I tried booting into Atmosphere, nothing. Blue screen still.

HaiKonofanDesu, 2024-06-11:
I already got the Bluescreen Fix Update and it fixes the initial Blue Screen.

Opening emuNand or SysNand once works, rebooting and (at leasst) switching to the other results in a blue screen. Have not yet tested if booting in the same one also results in a blue screen.

BlytheScythe, 2024-06-11:
I've replaced my reboot_to_payload.bin as well, still having the same issue.

mayt04, 2024-06-11:
Yeah,works fine with 6.1.1

Raph9213, 2024-06-11:
As @HaiKonofanDesu said, the first boot works fine for first time, and then do the "Blue Screen Code 01" panic after injecting again with Reboot Payload or with Studious Pancake, replacing manually atmosphere/reboot_payload.bin fixes the issue

CTCaer, 2024-06-11:
anyone that **does not have erista** should state that btw, like @lucasaf04

and as I said, if behavior changes on erista after a reboot (r2p), it's because of reboot_payload.bin

muxi1, 2024-06-11:
> @muxi1 on erista you need to replace reboot_payload.bin also if it is set to be replaced to hekate
I use a forwarder payload as reboot_payload.bin, which always loads the payload.bin (aka hekate_ctcaer_x.x.x.bin) from the SD card root.

HaiKonofanDesu, 2024-06-11:
> I already got the Bluescreen Fix Update and it fixes the initial Blue Screen. (using TegraRCMGUI 2.6)
> 
> Opening emuNand or SysNand once works, rebooting and (at leasst) switching to the other results in a blue screen. Have not yet tested if booting in the same one also results in a blue screen.

Using Hekate and Atmosphere

HaiKonofanDesu, 2024-06-11:
> As @HaiKonofanDesu said, the first boot works fine for first time, and then do the "Blue Screen Code 01" panic after injecting again with Reboot Payload or with Studious Pancake, replacing manually atmosphere/reboot_payload.bin fixes the issue

@Raph9213 with what file did you replace the reboot_payload.bin?

mayt04, 2024-06-11:
So we need to rename the latest hekate,bin to reboot_payload.bin  then we put it into the atmosphere folder,select replace and it should fix it on ersta?

CTCaer, 2024-06-11:
@muxi1 hekate uses the version to decide if it should launch an "updated" self. I didn't change that. So you are supposed to change all payloads that are hekate manually with updated hekate.
(e.g. Injector/loader payload, payload.bin, update.bin, reboot_payload.bin)

mayt04, 2024-06-11:
> So we need to rename the latest hekate_ctcaer.bin to reboot_payload.bin then we put it into the atmosphere folder,select replace and it should fix it on erista?

tried this,still doesnt works

CTCaer, 2024-06-11:
@Raph9213 It does work, I just didn't change the version on the fixed payload, so old and new are still 6.2.0.
It uses that to check if it needs updating, it doesn't randomly update it.
So the update.bin and reboot_payload.bin do not get updated because of that.

PreHeX, 2024-06-11:
> > > As @HaiKonofanDesu said, the first boot works fine for first time, and then do the "Blue Screen Code 01" panic after injecting again with Reboot Payload or with Studious Pancake, replacing manually atmosphere/reboot_payload.bin fixes the issue
> > 
> > 
> > @Raph9213 with what file did you replace the reboot_payload.bin?
> 
> With the hekate .bin payload in the zip file, the "hekate_ctcaer_6.2.0.bin" file, and renamed to "reboot_payload.bin"
> 
> It seems that the "Update Reboot 2 Payload" settings doesn't work anymore to "replace" the reboot payload for hekate?

does not work same error

HaiKonofanDesu, 2024-06-11:
@mayt04 @PreHeX You sure you have the updated Version?

The Method descriped by Raph worked for me.

Open atmosphere and delete reboot_payload
Copy the UPDATED (make sure it'S the fixed version) "hekate_ctcaer_6.2.0.bin" in the atmosphere folder and rename it again to "reboot_payload.bin"

Shutdown console and inject payload once more and it should work!

Ahh the same Version thin explains a lot, thanks for clarify!

mayt04, 2024-06-11:
> @mayt04 @PreHeX You sure you have the updated Version?
>
> The Method descriped by Raph worked for me.
>
> Open atmosphere and delete reboot_payload Copy the UPDATED (make sure it'S the fixed version) [...]

I did this metod and it fixed the issue,thanks everyone!

CTCaer, 2024-06-11:
@PreHeX 
[preHeX.zip](https://github.com/user-attachments/files/15788288/preHeX.zip)
Extract that update.bin to bootloader/ folder and replace one. Try to boot.
If that boots it means you didn't replace all relevant payloads.

No one else try that btw since I changed the internal version to 6.2.1 to bypass all that mumbo jumbo dancing around happening on eristas.

CTCaer, 2024-06-11:
> @mayt04 @PreHeX You sure you have the updated Version?
> 
> The Method descriped by Raph worked for me.
> 
> 1. Download the UPDATED files that include the Bluescreen Fix (hekate_ctcaer_6.2.0_Nyx_1.6.2, make sure your browser does not cache)
> 2. Shut down your Switch
> 3. Inject SD Card into PC
> 4. Open atmosphere folder and delete "reboot_payload.bin" file
> 5. Unzip the new "hekate_ctcaer_6.2.0_Nyx_1.6.2"
> 6. Open the Folder and copy the "hekate_ctcaer_6.2.0.bin" file
> 7. Paste it into the atmosphere folder and rename it again to "reboot_payload.bin"
> 8. Inject SD Card into PC, start Payload and you should be good to go
> 
> Shutdown console and inject payload once more and it should work!
> 
> Ahh the same Version thin explains a lot, thanks for clarify!

When I make sure that everything is fixed and all can boot, I will do a hotfix update as 6.2.1 end of this week or next.

mayt04, 2024-06-11:
Thank you for the patience guys

PreHeX, 2024-06-11:
> @PreHeX [preHeX.zip](https://github.com/user-attachments/files/15788288/preHeX.zip) Extract that update.bin to bootloader/ folder and replace one. Try to boot. If that boots it means you didn't [...]

no it still doesent work can i send u any logs somewhere or something because i even tried to turn on Logs but nothing happend...

CTCaer, 2024-06-11:
@PreHeX how you update the payloads? also grab the `sd:/bootloader/update.bin`, zip it and upload it here.

PreHeX, 2024-06-11:
> @PreHeX how you update the payloads? also grab the `sd:/bootloader/update.bin`, zip it and upload it here.

via Hekate -> USB Tools -> SD Card

[bootloader.zip](https://github.com/user-attachments/files/15788428/bootloader.zip)

CTCaer, 2024-06-11:
so you didn't update it. that has 6.2.0 as version.

PreHeX, 2024-06-11:
> so you didn't update it. that has 6.2.0 as version.

wait now im confused isnt that the latest ? or what do you mean "you didn´t update it"

mirivedan, 2024-06-11:
A clean install on 2 different sd cards gives me a black screen or screen crash both on OFW or CFW
[image]

BlytheScythe, 2024-06-11:
I've also double-checked whether everything's up-to-date and followed what [HaiKonofanDesu](https://github.com/HaiKonofanDesu) posted. Still blue screen. Here's my update.bin as well: 
[update.zip](https://github.com/user-attachments/files/15788549/update.zip)

PreHeX, 2024-06-11:
or should I wait until you fix it or 6.1.2 is released? I don't want to bother you now.

Callegari02, 2024-06-11:
> Un'installazione pulita su 2 diverse schede SD mi dà una schermata nera o un arresto anomalo dello schermo sia su OFW che su CFW ![IMG_2581](https://private-user-images.githubusercontent.com/41208329/338598800-037b64d4-af60-44d0-82b1-5b1d3c478e7a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxMTI0NDMsIm5iZiI6MTcxODExMjE0MywicGF0aCI6Ii80MTIwODMyOS8zMzg1OTg4MDAtMDM3YjY0ZDQtYWY2MC00NGQwLTgyYjEtNWIxZDNjNDc4ZTdhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjExVDEzMjIyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0OTBkMDVjYWYyNmY4YzdkMmYwYjc4ZWNmZmY2ZWJhNmQzYmFjNmIwOGUwNDhkZTY3OGI0OTM0YmI1MGJkZjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ni_st_CFsfFddszkaUbD3Sxt0BfNY5_tVEJr4z2w7hk)

It works like this for me too, if I put hetake 6.1.1 it works otherwise it doesn't work and it does what it does for you

mirivedan, 2024-06-11:
If i inject the new fusee.bin i can access atmosphere but it always crashes after 10 seconds

CTCaer, 2024-06-11:
Erista (V1)
@PreHeX I gave you a special one to try, to bypass all payload locations.
@BlytheScythe 
Anyway, both of you extract this and use payload.bin for injection (tegrasmash or loader)
[test.zip](https://github.com/user-attachments/files/15788663/test.zip)

Mariko (all):
@mirivedan @Callegari02 I said if it's a mariko you should state it.
Try this. Replace payload.bin and update.bin with that.
[payload.zip](https://github.com/user-attachments/files/15788683/payload.zip)

PreHeX, 2024-06-11:
Fix for every Erista Model/Switch v1

Launch latest Hekate

> Erista (V1) @PreHeX I gave you a special one to try, to bypass all payload locations. @BlytheScythe Anyway, both of you extract this and use payload.bin for injection (tegrasmash or loader) [...]

it works thank you very much

CTCaer, 2024-06-11:
@PreHeX so you didn't update one of the relevant payloads.
That `test.zip` basically had all relevant locations that the payload could be.

BlytheScythe, 2024-06-11:
@CTCaer I tied PreHeX's update.bin and it works. I've also just tried the test you've sent and it works as well.
I do find it hard to believe that I haven't updated all the necessary paylaods (even after a multiple times) but I'm glad it works properly now. Thank you so much!

PreHeX, 2024-06-11:
> @PreHeX so you didn't update one of the relevant payloads. That `test.zip` basically had all relevant locations that the payload could be.

Okay got it

CTCaer, 2024-06-11:
> @CTCaer I tried PreHeX's update.bin and it works. I've also just tried the test you've sent and it works as well. I do find it hard to believe that I haven't updated all the necessary paylaods (even after a multiple times) but I'm glad it works properly now. Thank you so much!

Well it's the same exact payload (the one in updated release), just copied pasted in every known location and filenames.
That's the issue with erista and every loader out there. Everyone used every single random place.
Anyway on the real update, that won't be needed since update.bin will have version 6.2.1.

mirivedan, 2024-06-11:
I replaced the old payload.bin and update.bin with the last payload.zip/bin and still doesn't work (Oled)
[image]

Callegari02, 2024-06-11:
> Ho sostituito i vecchi payload.bin e update.bin con l'ultimo payload.zip/bin e continua a non funzionare (Oled) ![IMG_2585](https://private-user-images.githubusercontent.com/41208329/338607400-a6eb162f-6f21-49a7-ba07-bb61b7bad1ff.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxMTM3MDMsIm5iZiI6MTcxODExMzQwMywicGF0aCI6Ii80MTIwODMyOS8zMzg2MDc0MDAtYTZlYjE2MmYtNmYyMS00OWE3LWJhMDctYmI2MWI3YmFkMWZmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjExVDEzNDMyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU1OTdlNjJlOWJlNGJkYzE3ZTMzYzI5ZTk1MDA1MWI1ZTQ3NmUxNzI2NjRjZmZkNTY5MDI2MjVhY2JjNWI2MjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.iXtZ4Rnfxo-v1-GkczrxXz_noNX4pm2e8ZWqTrX8PpI)

It says to me "pkg1 decryption failed
pkg1 corrupt?"

Callegari02, 2024-06-11:
> > Erista (V1)@PreHeXTe ne ho dato uno speciale da provare, per bypassare tutte le posizioni del carico utile.@BlytheScytheAd ogni modo, entrambi estraetelo e usate payload.bin per l'iniezione (tegrasmash o caricatore) [test.zip](https://github.com/user-attachments/files/15788663/test.zip)
> > Mariko (tutti):@mirivedan @Callegari02Ho detto che se è un mariko dovresti dichiararlo. Prova questo. Sostituisci payload.bin e update.bin con quello. [payload.zip](https://github.com/user-attachments/files/15788683/payload.zip)
> 
> Continua a non funzionare. emummc e sysmmc mostrano la schermata del problema tecnico ma in alto dice: decrittografia Pkg1 non riuscita. Impossibile avviare HOS. 17.0.0

Me too

mirivedan, 2024-06-11:
@lucasaf04  had also that error in one of my attempts

BlytheScythe, 2024-06-11:
@CTCaer I don't have much going on on my SD card, really. Pretty much the minimum stuff you need for CFW stuff and that's pretty much it. As for the payloads, I only use update.bin and reboot_payload.bin. I've noticed that you've added payload.bin to the root and hekate.bin in the payloads folder, none of which I usually use. Oh, and the Rekado's payloads which I update manually as well.

I usually update through Hekate's USB tools and just drag and drop, as per instructions. Therefore, update.bin should be up-to-date. As for the reboot_payload.bin, I made sure to update that one.

Maybe the issue lied with the versioning? As soon as you set it to 6.2.1, it magically worked. I'm not entirely familiar with the technical stuff but I'd hate myself to fail at copy-paste. :D 

Anyway, thanks for the hard work dealing with all of the issues. Keep up the good work!

CTCaer, 2024-06-11:
Mariko:
@lucasaf04 @mirivedan @Callegari02 
[payload.zip](https://github.com/user-attachments/files/15789036/payload.zip)

Callegari02, 2024-06-11:
> Mariko: @lucasaf04 @mirivedan @Callegari02 [payload.zip](https://github.com/user-attachments/files/15789036/payload.zip)
> 
> Inoltre, quale sdram avete ciascuno di voi? (id dram da Helate -> informazioni console -> informazioni hw)

my dram id is: 05: xynix H9HCNNNBKMMLXR - NEE 4GB

[13 more replies]

closed 2024-07-02 by CTCaer