CTCaer/hekate #971: Black screen after trying to boot hekate

Link019, 2024-06-11:
i’ve tried to boot hekate, but the screen remains black. i’m on 18.1 bc stupid system didn’t let me use my new micro sd card becuse it needed a system update

mightysashiman, 2024-06-11:
and it gets weirder for me: https://gbatemp.net/threads/18-1-0-ofw-does-not-boot-on-modded-oled.657003/

urherenow, 2024-06-12:
The SD thing was a mistake on your part. New SD cards over a certain size, ALWAYS come pre-formatted as exfat. Had you used nyx to format it to fat32, before booting your Switch, it would not have asked you to update to use the card. What you did, is force it to download the exfat drivers (but it won't do Only the drivers, if there is an available firmware update).

CTCaer, 2024-06-12:
@Link019 make sure you have v3 zip and replace all relevant bins you use as stated in https://github.com/CTCaer/hekate/issues/969
Other than that I can't control what happens after hekate launches HOS (which means you need to update incompatible sysmods/theme/etc on your own)

@mightysashiman you wiped your savedatas unofficially so I can't help on that.
Also ofw does not use hekate, stock does.

CTCaer, 2024-07-02:
v6.2.1 stable release is out

closed 2024-07-02 by CTCaer