CTCaer/hekate #973: Unable to launch Android from the new hekate update

Mika58Carca, 2024-06-12:
Hello
I am writing to you because indeed, since the new update of hekate recently, console switch refuses to boot on Android. 
It stays on the start screen and then nothing, forced to force the console off and then restart it. 
Android runs on the old hekate update ( hekate v6.1.1 & Nyx v1.6.1 )

CTCaer, 2024-06-12:
Since you actually checked to make sure there's a difference, what model you have?

Mika58Carca, 2024-06-12:
> Since you actually checked to make sure there's a difference, what model you have? And ofc which android flavor

To answer you, the model of my switch is a V1 erista. 
I'm running Android with lineageos 11 of memory.
Android works very well on hekate 6.1.1 but since the final version, it doesn't work anymore, I have to downgrade the version of hekate to make Android work again

maximo12345git, 2024-06-13:
> can also confirm that downgrading hekate to 6.1.1 fixed both this and #978 for me i have a XAW1000 switch (so v1 erista), am running lineageos 18.1 (android 11) and fedora 39 kde (did actually attempt an install after upgrading to 6.2.0, it didn't boot then but booted when i downgraded to 6.1.1)

But you did downgrade Custom firmware also? cause I treid to put back old Hekate but I've had an error on reload

Mika58Carca, 2024-06-13:
> But you did downgrade Custom firmware also? cause I treid to put back old Hekate but I've had an error on reload

In Hekate 6.1.1, atmosphere no longer works following the new Switch update 16.1.0 And Android works again. 
In Hekate, the latest current update, atmosphere works again with the new Switch 16.1.0 update but Android doesn't work anymore

RoblKyogre, 2024-06-13:
> But you did downgrade Custom firmware also? cause I treid to put back old Hekate but I've had an error on reload

i've actually yet to update to 18.1.0, so am still able to use cfw currently (6.1.1 doesn't support the latest firmware, 6.2.0 does)
will wait to upgrade until this gets fixed tho

urherenow, 2024-06-13:
I finally had the chance to try the latest hekate and Atmosphere on my OLED, and I haven’t been able to reproduce any problem posted about the latest update. Emummc -> reboot -> emummc, no problem. Launching lineage 18.1, no problems. My console is not updated to 18 though, because I simply don’t have the bandwidth to download it right now. My v1 switch is at home so no testing that until December…

CTCaer, 2024-06-13:
Fixed the issue. Get **_v4.zip**

Mika58Carca, 2024-06-13:
> Fixed the issue. Get **_v4.zip**

Thanks for your fix, everything works again with the latest version of hekateThanks for your fix, everything works again with the latest version of hekate ! 😎

josete2k, 2024-06-15:
Problems here with OLED.

Panic kernel with 6.2.0 v4.

This 6.1.1 compiled with 18.1.0 support works.

https://gbatemp.net/threads/rcm-payload-hekate-ctcaer-mod.502604/post-10438348

CTCaer, 2024-06-15:
@josete2k This is the wrong issue, move to https://github.com/CTCaer/hekate/issues/969
(You are supposed to replace payload.bin, although with 6.1.1 as it you can just place the bootloader folder now)

josete2k, 2024-06-15:
> @josete2k This is the wrong issue, move to #969 (You are supposed to replace payload.bin, although with 6.1.1 as it you can just place the bootloader folder now)

Ops sorry.

Just for clarification:

I always replace all bin files in my SD.. payload.bin, reboot_payload.bin, update.bin and hekate.bin (this last in bootloader/payloads folder).

But with v4 I get a kernel panic message in yellow letters, not blue o glitched screen... This is why I thought that this issue is the same I have.

CTCaer, 2024-06-15:
You talked about hos 18.1.0. Be specific without anything random
(because there's panic krnl for hos also and you could have used official 6.1.1 for that test.)

Nothing should affect L4T on mariko. Get the l4t panic log (L4T_panic.bin) since that's way after hekate.
And I suggest you to contact me in discord since it's easier to share files in succession there.

(Btw reboot_payload.bin is not used on mariko, as for the /payloads one, dunno why you use it)

josete2k, 2024-06-15:
I know that about reboot_payload.bin , but I have an erista V1 too and I share same files for both consoles.

L4T_panic.bin is present in the SD but it's empty... I've checked with winhex.

I'll retry again in a few hours.

Thanks in advance.

kevbenjam, 2024-06-17:
Thanks for trying to fix the issue with v4 @CTCaer. For me L4T is still not working via hekate 6.2.0 on Erista with Lineage OS 18.1 / 21. It hangs directly after showing "USB3 enabled" in the upper left corner.

CTCaer, 2024-06-17:
Since you see uboot printing a message (usb3 enabled), it's not that issue. The recent regression doesn't even reach uboot or TZ fw.
Either you didn't follow the android guide to the letter or your sd card or reader might have issues.

closed 2024-07-02 by CTCaer

Anonymous941, 2026-05-14:
I've been trying for days to read the pstore.  @CTCaer how did you mount it?  When I run `sudo mount -t pstore L4T_panic.bin mountpoint/`, it's just an empty folder.