CTCaer/hekate #978: Unable to boot Linux with Newest Hekate version

LuckyK65, 2024-06-12:
Hekate 6.2.0 v3 breaks linux booting as when you try to launch it, it gets frozen at the loading screen, after 10 minutes it still stays at the screen. Downgrading to 6.1.1 fixes the issue so something is wrong with the latest version.

iodn, 2024-06-12:
Same here, my Ubuntu version 22 was working fine before the update to Hekate 6.2.0 v3. After the update, Ubuntu seems to freeze on the boot screen.

CTCaer, 2024-06-12:
Assuming this is related to https://github.com/CTCaer/hekate/issues/973
I want model info.

LuckyK65, 2024-06-12:
Its the 2017 switch model

iodn, 2024-06-12:
> Assuming this is related to #973 I want model info.

Yeah, it looks like this is related indeed. 
My model is also a 1st gen switch (XAJ7002).

CTCaer, 2024-06-13:
Fixed the issue. Get **_v4.zip**

Fra2605, 2024-06-13:
Excuse me, I have the same problem even if I have updated hekate

Fra2605, 2024-06-13:
(V1 switch)

Fra2605, 2024-06-13:
Linux stuck on boot screen

azerty229, 2024-06-13:
v6.2.0 v4 no boot linux ubuntu 22 and 24 (erista v1 switch)

azerty229, 2024-06-14:
No boot kubuntu v24 on Switch erista v1 and hekate 6.2 V4

CTCaer, 2024-06-14:
@Fra2605 First of all don't spam, here's not a chat. Edit your posts.
@Fra2605 @azerty229 you are supposed to replace the payload you use to inject via PC or rcm loader or whatever.
So extract the bootloader folder to sd root and then inject the correct 6.2.0 hekate and not an older version as you do.

azerty229, 2024-06-14:
I have already changed all the bootloader directory I even formatted my sd card and I took hekate v6.2.0 v4 but no kubuntu boot

gitaccountme, 2024-06-26:
Hello
I'm confirming the same issue with Switch OLED model (linux is stuck on booting) I tried the following things with no avail :
- Upgrading to 6.2.0_V4.
- Formatting the SD card adding 6.2.0_V4  re-partitioning it again for fresh linux instantiation 
- Changing the SD card adding 6.2.0_V4.
- Changing from Ubuntu to Fedora 
- Downgrading to  hekate v6.1.1 

All resulted the same issue, note that Android and CFW boot are working fine

CTCaer, 2024-06-26:
@gitaccountme you are not confirming the same issue.
sd card and ddr200 issues are not a hekate issue and neither the same with above.
seek support to the correct place.

gitaccountme, 2024-06-26:
@CTCaer 

Hi sorry but I didn't get it how it's sd card and ddr200 issue?

My issue is:  "linux is stock on booting as when you try to launch it, it gets frozen at the loading screen, after 10 minutes it still stays at the screen"

CTCaer, 2024-06-26:
This not a hekate issue and support for that doesn't belong here. Seek help and the no ddr200 boot.scr in the l4s discord (find it in the wiki).

CTCaer, 2024-07-02:
v6.2.1 stable release is out

closed 2024-07-02 by CTCaer