CTCaer/hekate #979: Compilation challenges for new versions

ZeldaPken, 2024-06-13:
In the latest version, compiling the program becomes difficult, probably because you modify things like dependencies, etc., are there specific compilation steps and guidelines?

```$ make
Building build/hekate/start.o
bash: line 1: arm-none-eabi-gcc: command not found
make: *** [Makefile:154: build/hekate/start.o] Error 127```

error like this...

**Actually the arm-none-eabi-gcc I have configured correctly so I don't understand what else is missing**

ZeldaPken, 2024-06-13:
I used windowX86 and Msys2

urherenow, 2024-06-13:
Github isn’t a help forum. There are tutorials on gbatemp dedicated to helping people build their own. I’d ask what happens when you open msys2 and type arm-none-eabi-gcc (because it sounds like a path issue), but I’d rather you didn’t answer. Close this and seek help on temp…

ZeldaPken, 2024-06-13:
> Github 不是一个帮助论坛。有关于 gbatemp 的教程，专门帮助人们构建自己的教程。我会问当您打开 msys2 并键入 arm-none-eabi-gcc 时会发生什么（因为这听起来像是路径问题），但我宁愿您不回答。关闭它并寻求临时帮助...

GitHub isn't, but what is the issue?

Well, allow me to continue with my question, when I enter the command correctly, ```$ pacman -S gcc-arm-none-eabi
error: target not found: gcc-arm-none-eabi``` shows these two codes separately
```$ pacman -S gcc-arm-none-eabi
error: target not found: gcc-arm-none-eabi```

borntohonk, 2024-06-13:
> In the latest version, compiling the program becomes difficult, probably because you modify things like dependencies, etc., are there specific compilation steps and guidelines?
> 
> ```
> Building build/hekate/start.o
> bash: line 1: arm-none-eabi-gcc: command not found
> make: *** [Makefile:154: build/hekate/start.o] Error 127```
> 
> error like this...
> 
> Actually the arm-none-eabi-gcc I have configured correctly so I don't understand what else is missing
> ```

dependencies has not changed, latest commit compiles just fine, nothing has changed.

```
Debian GNU/Linux 12 (bookworm)

gcc-12-base/stable,now 12.2.0-14 amd64 [installed]
gcc-12/stable,now 12.2.0-14 amd64 [installed,automatic]
gcc-arm-none-eabi/stable,now 15:12.2.rel1-1 amd64 [installed]
gcc/stable,now 4:12.2.0-3 amd64 [installed]
```

closed 2024-06-13 by ZeldaPken