CTCaer/hekate #983: Enable Home & Screenshot buttons in USB Gamepad

benwa, 2024-06-17:
I see that you have the Home and Screenshot buttons disabled for USB Gamepad mode. Is there any reason as to why it can't be enabled?

https://github.com/CTCaer/hekate/blob/c321d3508cc757cf4d6a0800e6bc90838cc05830/bdk/usb/usb_gadget_hid.c#L290-L291

CTCaer, 2024-06-18:
This was done to limit the buttons at the standard number like all gamepads and also save 1 byte on the hid report.
Normally anything that supports dinput automatically maps these properly. For the rest, x360ce exists.

If there's popular demand, it can be changed. But only then, since I will have to re-read the usb hid spec and change the usb hid report descriptors.

benwa, 2024-06-18:
At least with Steam, those two buttons did not automatically map, so I can't natively bring up the menu.

Any chance to have it as an optional feature?

CTCaer, 2024-06-18:
You can't map them in host because they are not even reported, because of the reasons I said.

benwa, 2024-06-18:
Oh I understand, I was replying to the dinput comment

CTCaer, 2024-06-18:
The dinput comment means that normally software maps all these 8 buttons automatically.
Not the ones (cap/home) that are not reported and so they don't know that even exist.

benwa, 2024-06-18:
Ah, gotcha. I'll check a few Discord servers to see if anyone is using the feature and is interested in this. 

Thank you for the great work!